natario1/CameraView #747: if you set the pictureSize to a higher resolution then that of the previewStreamSize while still having the same aspect ratio, some of the picture won't be displayed.

Liduen, 2020-01-14:
### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: 2.6.0
- Camera engine used: camera2
- Reproducible in official demo app: no
- Device / Android version: Pixel 3 XL, API 29

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Use the setPictureSize method to set a resolution of 3840*2160
2. Use the setPreviewStreamSize to set a resolution of 1920*1080
    2.1. Note that both resolutions have an identical aspect ratio of 1.777778
3. Capture a photo while aligning the top left corner of your preview to a square object (like your computer monitor)
4. Observe the taken image.
5. Note that despite the object was aligned to the picture's corner of the preview, there's still some space between it and the actual corner of the taken picture

### Expected behavior
When having a pictureSize and a previewStreamSize of identical aspect ratios, the taken picture would match the image that's displayed in the preview.

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

        <com.otaliastudios.cameraview.CameraView
            android:id="@+id/camera_view"
            android:layout_width="wrap_content"
            android:layout_height="wrap_content"
            app:cameraAudio="off"
            app:cameraEngine="camera2"
            app:cameraPreview="texture"
            app:cameraExperimental="true"
            app:cameraUseDeviceOrientation="true"
            app:layout_constraintBottom_toBottomOf="parent"
            app:layout_constraintEnd_toEndOf="parent"
            app:layout_constraintStart_toStartOf="parent"
            app:layout_constraintTop_toTopOf="parent" />

### Screenshots
The image in preview before it's taken. Note how the window is aligned to the top left corner of the preview.
[image]

The image after it's taken (note the spacing between the opened window and the edge of the picture)
[image]

### Logs
Use `CameraLogger.setLogLevel(LEVEL_INFO)` to see all logs into LogCat.

Use `CameraLogger.registerLogger()` to export to file or crash reporting service.

### APK
Link to a Github repo where the bug is reproducible.

natario1, 2020-01-14:
If you force the view to fill the parent, what you see is not 1.778 anymore. You must use wrap_content

Liduen, 2020-01-14:
As you can see in the xml sample I’ve provided, the height and width of the
view are already set to wrap_content.

Liduen, 2020-01-19:
Just commenting again to make sure my previous comment was notified.

natario1, 2020-01-19:
It's not wrap_content as you can see from your first image - ConstraintLayout is forcing it to be match_parent due to your `layout_` attributes

Liduen, 2020-01-20:
When placing any view in ConstraintLayout, and adding the constraints I've added, the view's size isn't effected by the constraints (or is somehow converted to match_parent) unless the view's width and height are set to 0 (which it isn't).

example:
```
  <TextView
            android:layout_width="wrap_content"
            android:layout_height="wrap_content"
            android:background="#FF0000"
            android:text="POTATO"
            app:layout_constraintTop_toTopOf="parent"
            app:layout_constraintBottom_toBottomOf="parent"
            app:layout_constraintStart_toStartOf="parent"
            app:layout_constraintEnd_toEndOf="parent"/>
```

When I place this view in a ConstraintLayout, it looks like this (note the height, width and the constraints):

[image]

When I convert the layout_width to 0dp, it looks like this (now that the width is set to 0dp, the constraints dictate the width's behaviour):

[image]

The "size behaviour" is still dictated by the width and height which are set to "wrap_content". Is there any exception to this rule when using CameraView? should I just place the CameraView in a FrameLayout to avoid this scenario?

natario1, 2020-01-20:
Then I don't know @Liduen , maybe it looks match_parent because your screen is 1.77 as well, maybe you're not actually choosing 1.77 for preview (it's not guaranteed to be available), maybe ConstraintLayout is doing something weird, ... .

Liduen, 2020-01-23:
I tried placing the CameraView in a FrameLayout as such, while selecting a capture size of 2160x3840 and a preview size of 1080x1920,  still no success.

```
<FrameLayout
            android:layout_width="match_parent"
            android:layout_height="match_parent"
            android:background="#FF0000">

            <com.otaliastudios.cameraview.CameraView
                android:id="@+id/camera_view"
                android:layout_width="wrap_content"
                android:layout_height="wrap_content"
                android:layout_gravity="center"
                app:cameraAudio="off"
                app:cameraEngine="camera2"
                app:cameraFlash='@{viewModel.flashModeLd}'
                app:cameraUseDeviceOrientation="true" />

        </FrameLayout>
```

I'll try to remove the constraint layout altogether incase it's doing something weird

closed 2020-02-05 by natario1