natario1/CameraView #752: There was a problem recording the video

xionger0520, 2020-01-16:
Verison：2.6.0
Device : Huawei P20 Pro
Android:10.0
Mode :Camera2

Describe :
java.lang.IllegalStateException: Handler (android.os.Handler) {640a3bc} sending message to a Handler on a dead thread
        at android.os.MessageQueue.enqueueMessage(MessageQueue.java:618)
        at android.os.Handler.enqueueMessage(Handler.java:759)
        at android.os.Handler.sendMessageAtTime(Handler.java:708)
        at android.os.Handler.sendMessageDelayed(Handler.java:678)
        at android.os.Handler.post(Handler.java:407)
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler.post(WorkerHandler.java:164)
        at com.otaliastudios.cameraview.video.encoding.MediaEncoder.notify(MediaEncoder.java:239)
        at com.otaliastudios.cameraview.video.encoding.MediaEncoderEngine.notify(MediaEncoderEngine.java:191)
        at com.otaliastudios.cameraview.video.SnapshotVideoRecorder.onRendererFrame(SnapshotVideoRecorder.java:278)
        at com.otaliastudios.cameraview.preview.GlCameraPreview$Renderer.onDrawFrame(GlCameraPreview.java:237)
        at android.opengl.GLSurfaceView$GLThread.guardedRun(GLSurfaceView.java:1582)
        at android.opengl.GLSurfaceView$GLThread.run(GLSurfaceView.java:1281)

HBiSoft, 2020-01-18:
@xionger0520 
I have had so many crashes on Hauweii devices it's not even funny, see my comment here:
https://github.com/natario1/CameraView/issues/667#issuecomment-559843209

> This issue looks to be the same as #440 where it was closed because the logs were updated.
> 
> > Will improve error messages in #425 ... Without this there's not much we can do. If you debug yourself and find an issue, please reopen or open a new issue
> 
> **All of the crashes are on HUAWEI devices**, running Android 9.
> 
> I've had 97 crashes the past 7 days.
> 
> Here is a list of all the devices which experiences this issue:
> 
> * HUAWEI Honor 8X (HWJSN-H)
> * HUAWEI Y9 2018 (HWFLA-H)
> * HUAWEI P30 lite (HWMAR)
> * HUAWEI P30 Pro (HWVOG)
> * HUAWEI P20 Pro (HWCLT)
> * HUAWEI Mate 9 (HWMHA)

natario1, 2020-02-05:
This log warning that you reported can be safely ignored. You should also gracefully handle CameraExceptions about video recording failures in your app

closed 2020-02-05 by natario1