natario1/CameraView #755: App Crashes on rotation

Trust-Coder, 2020-01-17:
### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: 2.6.1
- Camera engine used: *camera2*
- Reproducible in official demo app: *don't know*
- Device / Android version: Galaxy Note10+ (d2q) (SM-N975U), Android 2.0.1, OS Version 10

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Go to Camera Screen
2. Rotate the device
3. NullPointerException rised

### Expected behavior
Shouldn't crash on rotation

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

```xml
                <com.otaliastudios.cameraview.CameraView
                    android:id="@+id/camera"
                    android:keepScreenOn="true"
                    android:layout_width="match_parent"
                    android:layout_height="wrap_content"
                    app:layout_constraintTop_toTopOf="parent"
                    app:layout_constraintBottom_toBottomOf="parent"
                    app:layout_constraintEnd_toEndOf="parent"
                    app:layout_constraintStart_toStartOf="parent"
                    app:cameraGrid="draw3x3"
                    app:cameraPlaySounds="false"
                    app:cameraMode="picture"
                    app:cameraHdr="on"
                    app:cameraPictureSnapshotMetering="true"
                    app:cameraWhiteBalance="auto"

                    app:cameraEngine="camera2"
                    app:cameraAutoFocusMarker="@string/cameraview_default_autofocus_marker"
                    app:cameraPreview="texture"/>
```

### Logs
java.lang.NullPointerException: Attempt to invoke virtual method 'com.otaliastudios.cameraview.frame.ByteBufferFrameManager com.otaliastudios.cameraview.engine.Camera1Engine.getFrameManager()' on a null object reference

java.lang.NullPointerException: Attempt to invoke virtual method 'com.otaliastudios.cameraview.frame.ByteBufferFrameManager com.otaliastudios.cameraview.engine.Camera1Engine.getFrameManager()' on a null object reference
	at com.otaliastudios.cameraview.picture.Snapshot1PictureRecorder$1.onPreviewFrame(Snapshot1PictureRecorder.java:84)
	at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1221)
	at android.os.Handler.dispatchMessage(Handler.java:107)
	at android.os.Looper.loop(Looper.java:237)
	at android.os.HandlerThread.run(HandlerThread.java:67)

natario1, 2020-01-17:
Thanks! Would you like to fix this with a PR? I think we should just remove the method `Snapshot1PictureRecorder.dispatchResult()`.

To make it more elegant you could also make the 4 variables `final`.

android-dataticket, 2020-11-02:
I get this randomly on version 2.6.4 - any update on the fix?

android-dataticket, 2020-11-10:
this is still happening frequently on version 2.6.4, any hints to fix?

Landi29, 2021-01-09:
@natario1  I have a made a solution and want to create a PR, however I get an permission denied when I want to push my branch.