natario1/CameraView #758: Multiple bugs on Huawei devices

HBiSoft, 2020-01-18:
### Describe the bug
There are a lot of issues opened (and closed due to stale) that have errors on Huawei devices. Thus, I decided to open a new one to hopefully get some attention to the issue.

- CameraView version: *2.6.1*
- Camera engine used: *both (But mainly camera1)*
- Reproducible in official demo app: *Do not own a Huawei device*
- Device / Android version: 
*HUAWEI Honor 8X (HWJSN-H)
HUAWEI Y9 2018 (HWFLA-H)
HUAWEI P30 lite (HWMAR)
HUAWEI P30 Pro (HWVOG)
HUAWEI P20 Pro (HWCLT)
HUAWEI Mate 9 (HWMHA)
ALL RUNNING ANDROID 9*

### List of issues:
- #752
- #742 
- #701
- #682
- #667 

I do not have a Huawei device to test with, so I can only provide the log provided in the Play Store (Crashlytics does not catch this):

[image]

I wish I could provide more information about this issue, but this is all I have at this moment.

cleancontrol, 2020-01-22:
I got this exception : on huawei MediaPad T3 7 (Tablet)

```
2020-01-22 21:30:30.779 22037-23316/com.naqicontrol.clean W/MessageQueue: Handler (android.os.Handler) {37d621b} sending message to a Handler on a dead thread
    java.lang.IllegalStateException: Handler (android.os.Handler) {37d621b} sending message to a Handler on a dead thread
        at android.os.MessageQueue.enqueueMessage(MessageQueue.java:603)
        at android.os.Handler.enqueueMessage(Handler.java:653)
        at android.os.Handler.sendMessageAtTime(Handler.java:622)
        at android.os.Handler.sendMessageDelayed(Handler.java:592)
        at android.os.Handler.post(Handler.java:345)
        at android.hardware.camera2.impl.CameraDeviceImpl.close(CameraDeviceImpl.java:991)
        at com.otaliastudios.cameraview.k.b.Q(SourceFile:740)
        at com.otaliastudios.cameraview.k.d$i.call(SourceFile:414)
        at com.otaliastudios.cameraview.k.d$i.call(SourceFile:411)
        at com.otaliastudios.cameraview.k.l.c$b.call(SourceFile:69)
        at com.otaliastudios.cameraview.k.l.c$b.call(SourceFile:60)
        at com.otaliastudios.cameraview.k.l.a$b.onComplete(SourceFile:100)
        at com.google.android.gms.tasks.zzj.run(SourceFile:4)
        at com.otaliastudios.cameraview.p.f.g.c(SourceFile:137)
        at com.otaliastudios.cameraview.p.f.g$b.execute(SourceFile:112)
        at com.google.android.gms.tasks.zzi.onComplete(SourceFile:10)
        at com.google.android.gms.tasks.zzr.zza(SourceFile:19)
        at com.google.android.gms.tasks.zzu.trySetResult(SourceFile:94)
        at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(SourceFile:10)
        at com.otaliastudios.cameraview.k.l.a$b$a.onComplete(SourceFile:116)
        at com.otaliastudios.cameraview.k.l.a$d.run(SourceFile:195)
        at com.otaliastudios.cameraview.p.f.g.c(SourceFile:137)
        at com.otaliastudios.cameraview.k.l.a.b(SourceFile:192)
        at com.otaliastudios.cameraview.k.l.a.a(SourceFile:32)
        at com.otaliastudios.cameraview.k.l.a$b.onComplete(SourceFile:101)
        at com.google.android.gms.tasks.zzj.run(SourceFile:4)
        at com.otaliastudios.cameraview.p.f.g.c(SourceFile:137)
        at com.otaliastudios.cameraview.p.f.g$b.execute(SourceFile:112)
        at com.google.android.gms.tasks.zzi.onComplete(SourceFile:10)
        at com.google.android.gms.tasks.zzr.zza(SourceFile:19)
        at com.google.android.gms.tasks.zzu.trySetResult(SourceFile:94)
        at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(SourceFile:10)
        at com.otaliastudios.cameraview.k.l.a$b$a.onComplete(SourceFile:116)
        at com.otaliastudios.cameraview.k.l.a$d.run(SourceFile:195)
        at com.otaliastudios.cameraview.p.f.g.c(SourceFile:137)
        at com.otaliastudios.cameraview.k.l.a.b(SourceFile:192)
        at com.otaliastudios.cameraview.k.l.a.a(SourceFile:32)
        at com.otaliastudios.cameraview.k.l.a$b.onComplete(SourceFile:101)
        at com.otaliastudios.cameraview.k.l.a$d.run(SourceFile:195)
        at android.os.Handler.handleCallback(Handler.java:761)
        at android.os.Handler.dispatchMessage(Handler.java:98)
        at android.os.Looper.loop(Looper.java:156)
        at android.os.HandlerThread.run(HandlerThread.java:61)
2020-01-22 21:30:30.779 22037-23316/com.naqicontrol.clean I/d: onStopEngine: Clean up. Released camera.
2020-01-22 21:30:30.779 22037-23316/com.naqicontrol.clean I/d: onStopEngine: Aborting actions.
2020-01-22 21:30:30.780 22037-23316/com.naqicontrol.clean W/d: onStopEngine: Returning.
2020-01-22 21:30:30.780 22037-23316/com.naqicontrol.clean I/a: ENGINE << OFF - Finished.
2020-01-22 21:30:30.781 22037-22037/com.naqicontrol.clean E/d: EXCEPTION: Got CameraException. Dispatching to callback.
```

HBiSoft, 2020-03-19:
Bump stale

closed 2020-04-16 by stale[bot]

HBiSoft, 2020-05-19:
Bump stale