natario1/CameraView #761: Fatal Exception: java.lang.RuntimeException setParameters failed

daxeshvekariya, 2020-01-21:
Device: Samsung Galaxy On5 
Lib Version: Latest 
OS : Android 7

Fatal Exception: java.lang.RuntimeException: setParameters failed
       at android.hardware.Camera.native_setParameters(Camera.java)
       at android.hardware.Camera.setParameters(Camera.java:1940)
       at com.otaliastudios.cameraview.engine.Camera1Engine.applyZoom(Camera1Engine.java:613)
       at com.otaliastudios.cameraview.engine.Camera1Engine.applyAllParameters(Camera1Engine.java:452)
       at com.otaliastudios.cameraview.engine.Camera1Engine.onStartEngine(Camera1Engine.java:168)
       at com.otaliastudios.cameraview.engine.CameraEngine$5.call(CameraEngine.java:390)
       at com.otaliastudios.cameraview.engine.CameraEngine$5.call(CameraEngine.java:383)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:69)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:60)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:100)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
       at android.os.Handler.handleCallback(Handler.java:751)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:154)
       at android.os.HandlerThread.run(HandlerThread.java:61)

natario1, 2020-02-05:
Thanks for reporting but we are not going to fix any issue about Camera1 anymore, sorry. Your error can happen on specific devices or when you are not managing the camera state correctly.

closed 2020-02-05 by natario1

daxeshvekariya, 2020-02-06:
But I have just used as per your documentation describe. please guide me on how I manage the camera state correctly using your library? I haven't found any steps in docs.