natario1/CameraView #762: NullPointerException in onRendererFilterChanged

usmanrana07, 2020-01-21:
### Describe the bug
When i want to capture an image for first time and as i tap Cam icon after allow the permissions app is being crashed.

- CameraView version: *2.6.1*
- Device / Android version: *Samsung A30, API 28*

### Logs

> Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'void com.otaliastudios.cameraview.internal.d.c.a(com.otaliastudios.cameraview.l.b)' on a null object reference
>        at com.otaliastudios.cameraview.picture.SnapshotGlPictureRecorder.onRendererFilterChanged + 134(SnapshotGlPictureRecorder.java:134)
>        at com.otaliastudios.cameraview.picture.SnapshotGlPictureRecorder$1.onRendererFilterChanged + 98(SnapshotGlPictureRecorder.java:98)
>        at com.otaliastudios.cameraview.preview.GlCameraPreview$2.run + 305(GlCameraPreview.java:305)
>        at android.opengl.GLSurfaceView$GLThread.guardedRun + 1490(GLSurfaceView.java:1490)
>        at android.opengl.GLSurfaceView$GLThread.run + 1262(GLSurfaceView.java:1262)

closed 2020-02-19 by stale[bot]

brnogz, 2020-03-02:
@natario1 I have the same issue, do I need to create new issue ?  There is nothing I can add but it is trending crash on crashlytics.

natario1, 2020-03-02:
I think this might be fixed already @brnogz . If the issue persists with next version feel free to open a new issue!

VolodymyrMachekhin, 2020-04-15:
Got this issue in 2.6.2. Also what I see in logs is `app passed NULL surface`. 
Issue appears only after granting permissions.