natario1/CameraView #767: CameraView is fine. Video is blurry when we move camera or the person in video moves his hands.

ayushkul, 2020-01-29:
```
<com.otaliastudios.cameraview.CameraView
            android:id="@+id/camera"
            android:layout_width="match_parent"
            android:layout_height="match_parent"
            android:layout_gravity="center"
            android:keepScreenOn="true"
            app:cameraAudio="on"
            app:cameraAutoFocusMarker="@string/cameraview_default_autofocus_marker"
            app:cameraGestureScrollVertical="exposureCorrection"
            app:cameraEngine="camera1"
            app:cameraExperimental="false"
            app:cameraFacing="back"
            app:cameraFlash="auto"
            app:cameraGestureLongTap="none"
            app:cameraGesturePinch="zoom"
            app:cameraGestureTap="autoFocus"
            app:cameraGrid="off"
            app:cameraMode="video"
            app:cameraVideoBitRate="0"
            app:cameraVideoCodec="h264"
            app:cameraPlaySounds="true"
            app:cameraPreview="glSurface"
            app:cameraVideoMaxDuration="0"
            app:cameraVideoMaxSize="0">
```

natario1, 2020-01-30:
You likely need a higher `videoBitRate`

closed 2020-02-05 by natario1

ayushkul, 2020-02-20:
Can you suggest the `videoBitRate` ?

SreeniMannem, 2020-06-30:
Can we opt for Adaptive Bitrate by using this CameraView library? If yes, how can we implement this. Thanks in advance.