natario1/CameraView #782: Is it possible to add horizontal red line in center of screen?

alexei-28, 2020-02-14:
In my Android project:

 ```

implementation "com.otaliastudios:cameraview:2.6.0"

 <androidx.constraintlayout.widget.ConstraintLayout
                android:id="@+id/containerCameraView"
                android:layout_width="0dp"
                android:layout_height="0dp"
                android:layout_margin="@dimen/min_height"
                app:layout_constraintBottom_toTopOf="@+id/scanHinttextView"
                app:layout_constraintEnd_toEndOf="parent"
                app:layout_constraintStart_toStartOf="parent"
                app:layout_constraintTop_toTopOf="parent">

                <com.otaliastudios.cameraview.CameraView
                    android:id="@+id/cameraView"
                    android:layout_width="0dp"
                    android:layout_height="0dp"
                    android:keepScreenOn="true"
                    app:cameraAudio="off"
                    app:cameraMode="picture"
                    app:cameraPictureFormat="jpeg"
                    app:cameraRequestPermissions="false"
                    app:layout_constraintBottom_toBottomOf="parent"
                    app:layout_constraintEnd_toEndOf="parent"
                    app:layout_constraintStart_toStartOf="parent"
                    app:layout_constraintTop_toTopOf="parent" />

            </androidx.constraintlayout.widget.ConstraintLayout>
```

Nice it's work fine:

[image]

But I need to show horizontal red line in center of screen. Smt like this:

[image]

Is it possible?

yuripourre, 2020-03-05:
@alexei-28 should be easy to do it, you could create a relative layout or a frame layout and add CameraView and your overlay (the red line) to it.

```
<RelativeLayout>
  <com.otaliastudios.cameraview.CameraView>
  ...
  </com.otaliastudios.cameraview.CameraView>
  <View>
  <!--my red line-->
  </View>
```

A more complete answer: https://stackoverflow.com/a/7519248/7030976

Please tell me if you need more details

alexei-28, 2020-03-05:
> @alexei-28 should be easy to do it, you could create a relative layout or a frame layout and add CameraView and your overlay (the red line) to it.
> 
> ```
> <RelativeLayout>
>   <com.otaliastudios.cameraview.CameraView>
>   ...
>   </com.otaliastudios.cameraview.CameraView>
>   <View>
>   
>   </View>
> </RelativeLayout>
> ```
> 
> A more complete answer: https://stackoverflow.com/a/7519248/7030976
> 
> Please tell me if you need more details

It's not help. Because red line IS NOT STATIC. It's dynamic. Open any standard iPhone camera and you will see this red line which is constantly moving.

yuripourre, 2020-03-05:
@alexei-28 you haven't mentioned it, I never had an iPhone so I had no idea about the behavior of this red line.

In this case, I will assume that this line is updated given the smartphone angle. So, you could include your custom view (that shows the red line) and in your Activity/Fragment you will request the accelerometer sensor and when the value is updated, you could rotate the view.

You should be able to do this in any Android App, it is definitely not related/specific to CameraView's library.

alexei-28, 2020-03-06:
Red line must blink

closed 2020-04-02 by stale[bot]