natario1/CameraView #783: Is it possible to show square in the center?

alexei-28, 2020-02-14:
android project:

`implementation "com.otaliastudios:cameraview:2.6.0"`

Is is possible to do like this?

[image]

yuripourre, 2020-03-05:
@alexei-28 yes, it is possible. You could create a RelativeLayout or a FrameLayout and add CameraView and then your overlay (the square) to it. As overlay you could create a vertical linearlayout, for example, and divide into three parts:

- The upper part would be transparent (black tint).
- The middle part could be another LinearLayout but this time horizontal, weighted and divided into three (transparent, your image, transparent), remember to use the same weight into transparent parts
- The bottom part would be transparent (black tint) the same as the upper.

```
<RelativeLayout>
  <com.otaliastudios.cameraview.CameraView>
  ...
  </com.otaliastudios.cameraview.CameraView>
  <View>
  <!--my overlay-->
  </View>
</RelativeLayout>
```

You should be able to do that in any Android App. It is not related/specific to CameraView library
A more complete answer: https://stackoverflow.com/a/7519248/7030976

Please tell me if you need more details

closed 2020-04-01 by stale[bot]

theanilpaudel, 2022-09-27:
Is there any way to capture only the part that is visible in the overlay square?