natario1/CameraView #788: Flash unreliable for Samsung Galaxy S7 Camera2 API

gunnarvartdal, 2020-02-27:
### Describe the bug

- CameraView versions: `2.6.0`, `2.6.1`, latest (`aa95d19`)
- Camera engine used: *camera2*
- Reproducible in official demo app: *yes*
- Device / Android version: *Two separate Samsung Galaxy S7, Android 8.0, API 26*

Flash timing is off or unreliable on Samsung Galaxy S7 when using camera2 api. Torch-metering in this process is also unreliable. And also whether flash fires or not, is also unreliable. In other words, any combination of:
- flash fired but timing issue caused the image to be captured either right before or after the flash.
- metering performed, but no flash fired
- no metering performed, no flash fired

In all cases, the flash setting was set to "on". This affects both jpeg and dng, but the timing issue is much more prevalent for `dng`. It is exceedingly difficult to get a capture using `dng`, where the image is illuminated by the flash, even if the flash fired at seemingly correct time.

We've tested with two Samsung Galaxy S7, and on a Pixel XL. The issue rarely happened on the Pixel XL. Using camera1 API, there were no flash issues, though then of course limited to jpeg.

We've also tried changing the capture process by overriding `onAutoFocusEnd` where `camera.takePicture()` is called. And replacing the call to `camera.takePicture()` in `capturePicture()` to a call to `camera.startAutoFocus()`. Doing this, the flash was far more reliable, but mostly just for `jpeg` (about 90% success rate). However, acquiring focus in this way takes about 2-4 seconds, without the camera actually changing focus during this time. For `dng`, the timing issue is still present, with a success rate of around 10%.

### To Reproduce
1. Run the demo app as is.
2. Take pictures of a reflective surface, e.g. spoon.
3. One of:
- Observe lack of flash in images where flash fired.
- Observe no flash firing, even though flash is `on`.
### Expected behavior
More or less the `camera1` behavior. If flash is `on`, it means an image is captured where the flash was fired.

natario1, 2020-03-03:
Hi @gunnarvartdal - could you pull the latest master version and send me logs from the demo app? I don't have this specific device.

imanism, 2020-05-07:
Facing similar issue on Samsung Galaxy Note 10+. While I'm trying to figure out a solution, I resorted to using TORCH instead which works.

closed 2020-07-12 by natario1