natario1/CameraView #792: Occasionally getting takePicture failed crash

muammil, 2020-03-05:
### Describe the bug
Irrespective of the android version and device model I am getting this **Full1PictureRecorder.java line 44** **com.otaliastudios.cameraview.picture.Full1PictureRecorder.take** crash occasionally.
- CameraView version: *2.6.1*
- Camera engine used: *Not specified engine in xml or in code*
- Reproducible in official demo app: *no*
- Device / Android version: *Got crash report from all android versions except android 10 and reports are mainly from brands like Xiaomi, Oppo, Samsung, Vivo, etc.*

### To Reproduce
So far I wasn't able to reproduce it, reporting this based on the crash reports in crashlytics

### Expected behavior
1. My app minimum sdk version is 21, so I wish to know will this library work on camera2 automatically, I mean without specifying in code or xml?
2. If I specify camera engine to camera2 in xml do I have to add the app:cameraExperimental="true" in xml? and what this tag actually do, will it break any other thing?

I believe the crash is happening in camera1 engine, if I set it to camera2 will it resolve my issue?

### XML layout
```xml
<CameraView
      android:keepScreenOn="true"
      android:layout_width="wrap_content"
      android:layout_height="wrap_content"
      android:layout_gravity="center"
      app:cameraFacing="back"
      app:cameraFlash="auto"
      app:cameraPictureFormat="jpeg"
      app:cameraAudio="off"
      app:cameraMode="picture"
      app:cameraUseDeviceOrientation="false"
      app:cameraPictureSizeMinWidth="640"
      app:cameraPictureSizeMinHeight="480"
      app:cameraPictureSizeMaxWidth="960"
      app:cameraPictureSizeMaxHeight="720" />
```

### Screenshots
If applicable, add screenshots to help explain your problem.

### Logs
`Fatal Exception: java.lang.RuntimeException: takePicture failed
       at android.hardware.Camera.native_takePicture(Camera.java)
       at android.hardware.Camera.takePicture(Camera.java:1935)
       at com.otaliastudios.cameraview.picture.Full1PictureRecorder.take(Full1PictureRecorder.java:44)
       at com.otaliastudios.cameraview.engine.Camera1Engine.onTakePicture(Camera1Engine.java:347)
       at com.otaliastudios.cameraview.engine.CameraBaseEngine$3.run(CameraBaseEngine.java:498)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$3.run(CameraStateOrchestrator.java:100)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:75)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:72)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:100)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
       at android.os.Handler.handleCallback(Handler.java:790)
       at android.os.Handler.dispatchMessage(Handler.java:99)
       at android.os.Looper.loop(Looper.java:192)
       at android.os.HandlerThread.run(HandlerThread.java:65)`

natario1, 2020-03-05:
You need to add cameraExperimental=true, and no it should not break anything!

muammil, 2020-03-06:
okay, I will add 
`app:cameraEngine="camera2"
 app:cameraExperimental="true"`
in the xml, and hope I will not get **takePicture** crash again

closed 2020-03-06 by natario1