natario1/CameraView #797: Preview issue with camera2 / Picture Mode / Back Camera on Pixel 4

vork, 2020-03-10:
### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: *version number*
- Camera engine used: *camera2*
- Reproducible in official demo app: *yes*
- Device / Android version: *Pixel 4, API 29*

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Just start the app and observe the preview screen

Somehow the image is shown multiple time stacked horizontally. This only happens when using camera2, picture mode and the backfacing camera. 

### Expected behavior
A regular preview image.

### Screenshots

[image]

### Logs

[Log](https://pastebin.com/NSe9S0GP)

natario1, 2020-03-10:
Thanks! What about the version number? Does this happen in latest version? And what about 2.6.1?

vork, 2020-03-10:
The demo is from the 2.6.2 and I cannot reproduce it on 2.6.1.

Btw. this does not happen on a Pixel 3a on API 29 and a Moto G7 on API 28

natario1, 2020-03-10:
From your logs I just see that camera configuration failed. Can you try any of the steps mentioned in #607 ?

- Reduce picture size: call for example cameraView.setPictureSize(SizeSelectors.smallest())
- Reduce preview size: call for example cameraView.setPreviewStreamSize(SizeSelectors.smallest())

closed 2020-04-06 by stale[bot]

HiDream, 2020-06-09:
> The demo is from the 2.6.2 and I cannot reproduce it on 2.6.1.
> 
> Btw. this does not happen on a Pixel 3a on API 29 and a Moto G7 on API 28

have solved this problem? i met this too.