natario1/CameraView #805: Disable restart camera after change mode (photo/video)

Namazed, 2020-03-23:
### Problem to be addressed
In my app I need fast switch between mode photo and video, because is more comfortable for users, who use camera in my app.

I looked at source code where we switch mode and saw that on Engine.Camera1 we can stop restart camera, because it's not need. On Engine.Camera2 do it it's not simple, because you don't create video and photo recorder for one request, but we can use in one moment photo and video request.

### Describe the solution you'd like
I will create PR with solution for Engine.Camera1, because I think I understood how this work. 
About Camera2, I tried use both photo and video request in Engine.Camera2 and system threw exception, that telling that request bad configured. Maybe you know how we can do it on Engine.Camera2.

### Additional context
This will use flag, like requestPermissions.

Namazed, 2020-04-13:
Up

Terenfear, 2020-05-03:
Up

Terenfear, 2020-05-24:
Up

Namazed, 2020-06-13:
Up

Namazed, 2020-07-05:
up

Namazed, 2020-07-27:
up

Namazed, 2020-08-17:
up

closed 2020-09-13 by stale[bot]