natario1/CameraView #807: suppost disable auto rotation video when takeVideoSnapshot

khanhtonbsv, 2020-03-23:
I want to keep the video orientation vertically even if the device is rotated, can you add this feature?

natario1, 2020-03-24:
You can use cameraUseDeviceOrientation=false .

khanhtonbsv, 2020-03-24:
> You can use cameraUseDeviceOrientation=false .
But when I turned the device horizontally and recorded, the result received a rotation of 0 and the video was horizontal. In my case I wanted to lock it, even though the device is horizontal, the video orientation is still vertical. Because I need to merge several videos into one so I need them to be the same size

natario1, 2020-03-24:
I don't think it's the camera responsibility to set the orientation to an arbitrary value.

You can use https://github.com/natario1/Transcoder to merge videos and rotate them at the same time, even if they have different sizes.

khanhtonbsv, 2020-03-24:
Thank you for your help

closed 2020-03-24 by natario1