natario1/CameraView #808: Live Preview all filters

msroid, 2020-03-24:
I want to have a preview of any available filter so that user can see it before selecting.
I want to do something like this at the bottom of  the image:

[https://evolutionmentor.com/wp-content/uploads/2018/02/instagram-2.jpg](url)

this is for image but I want live preview  of camera.

I appreciate any help

best regards

natario1, 2020-03-24:
We do not support multiple instances of CameraView. Adding multiple previews is complex so I can't help here, sorry. If you represent a company feel free to send an e-mail.

msroid, 2020-03-24:
Any idea how can I apply those filters on a bitmap?
I have no experience of working with opengl!
If there's a way for doing this I can get the preview bitmap and then apply the filter on it.
best regards

natario1, 2020-03-24:
You can't apply them directly to Bitmaps unfortunately.

closed 2020-03-26 by msroid