natario1/CameraView #810: Video Record crashs in some devices

luangs7, 2020-03-24:
Video record crashs in some devices, most of Motorola`s devices. 
I can see the log only in crashlytics, I dont have a device in my hands with this problem.

- CameraView version: 2.4.0
- Reproducible in official demo app: not sure
- Device / Android version: Moto G5

LOG:

```
Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'android.hardware.camera2.CaptureRequest android.hardware.camera2.impl.CameraDeviceImpl$CaptureCallbackHolder.getRequest(int)' on a null object reference
       at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.onResultReceived(CameraDeviceImpl.java:1758)
       at android.hardware.camera2.legacy.CameraDeviceUserShim$CameraCallbackThread$CallbackHandler.handleMessage(CameraDeviceUserShim.java:316)
       at android.os.Handler.dispatchMessage(Handler.java:102)
       at android.os.Looper.loop(Looper.java:154)
       at android.os.HandlerThread.run(HandlerThread.java:61)

```

natario1, 2020-03-24:
I recommend updating to latest version. There's also nothing we can do without logs

closed 2020-04-23 by stale[bot]

luangs7, 2020-05-13:
I have updated to the last version and still have problems with it