natario1/CameraView #812: Crash Fatal Exception: java.lang.RuntimeException

yankelandroid, 2020-03-30:
### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: *2.0.0-beta02*
- Camera engine used: *camera1*
- Reproducible in official demo app: *no*
- Device / Android version: *Pixel, API 28*

### To Reproduce
Do not know i receive this error a lot of recurence from firebase

### Expected behavior
A clear and concise description of what you expected to happen.

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

```xml
<com.otaliastudios.cameraview.CameraView
        android:id="@+id/camera"
        android:layout_width="match_parent"
        android:layout_height="match_parent"
        android:layout_gravity="center"
        android:keepScreenOn="true"
        app:cameraAudio="on"
        app:cameraExperimental="true"
        app:cameraFlash="off"
        app:cameraGestureLongTap="none"
        app:cameraGesturePinch="zoom"
        app:cameraGestureScrollHorizontal="exposureCorrection"
        app:cameraGestureScrollVertical="none"
        app:cameraGestureTap="focusWithMarker"
        app:cameraGrid="off"
        app:cameraPictureSizeAspectRatio="9:16"
        app:cameraMode="picture"
        app:cameraPlaySounds="true" />
```

### Screenshots
If applicable, add screenshots to help explain your problem.

### Logs
Caused by java.lang.StackOverflowError: stack size 8MB
       at android.os.Parcel.readString(Parcel.java:2020)
       at android.text.TextUtils$1.createFromParcel(TextUtils.java:741)
       at android.text.TextUtils$1.createFromParcel(TextUtils.java:733)
       at android.content.pm.PackageItemInfo.<init>(PackageItemInfo.java:735)
       at android.content.pm.ApplicationInfo.<init>(ApplicationInfo.java:1796)
       at android.content.pm.ApplicationInfo.<init>(ApplicationInfo.java:60)
       at android.content.pm.ApplicationInfo$1.createFromParcel(ApplicationInfo.java:1787)
       at android.content.pm.ApplicationInfo$1.createFromParcel(ApplicationInfo.java:1785)
       at android.content.pm.PackageInfo.<init>(PackageInfo.java:513)
       at android.content.pm.PackageInfo.<init>(PackageInfo.java:31)
       at android.content.pm.PackageInfo$1.createFromParcel(PackageInfo.java:492)
       at android.content.pm.PackageInfo$1.createFromParcel(PackageInfo.java:489)
       at android.content.pm.IPackageManager$Stub$Proxy.getPackageInfo(IPackageManager.java:3396)
       at android.app.ApplicationPackageManager.getPackageInfoAsUser(ApplicationPackageManager.java:261)
       at android.app.ApplicationPackageManager.getPackageInfo(ApplicationPackageManager.java:239)
       at com.otaliastudios.cameraview.CameraView.checkPermissionsManifestOrThrow(CameraView.java:651)
       at com.otaliastudios.cameraview.CameraView.checkPermissions(CameraView.java:624)
       at com.otaliastudios.cameraview.CameraView.open(CameraView.java:606)
       at com.app.Activity.onRequestPermissionsResult(CameraActivity.java:921)
       at android.app.Activity.requestPermissions(Activity.java:4587)
       at com.otaliastudios.cameraview.CameraView.requestPermissions(CameraView.java:1464)
       at com.otaliastudios.cameraview.CameraView.checkPermissions(CameraView.java:636)
       at com.otaliastudios.cameraview.CameraView.open(CameraView.java:606)
       at com.app.Activity.onRequestPermissionsResult(CameraActivity.java:921)
       at android.app.Activity.requestPermissions(Activity.java:4587)
       at com.otaliastudios.cameraview.CameraView.requestPermissions(CameraView.java:1464)
       at com.otaliastudios.cameraview.CameraView.checkPermissions(CameraView.java:636)
       at com.otaliastudios.cameraview.CameraView.open(CameraView.java:606)
[... 50 lines cut ...]
       at androidx.lifecycle.ReportFragment.dispatch(ReportFragment.java:123)
       at androidx.lifecycle.ReportFragment.onResume(ReportFragment.java:90)
       at android.app.Fragment.performResume(Fragment.java:2568)
       at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:1341)
       at android.app.FragmentManagerImpl.moveFragmentToExpectedState(FragmentManager.java:1576)
       at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:1637)
       at android.app.FragmentManagerImpl.dispatchMoveToState(FragmentManager.java:3050)
       at android.app.FragmentManagerImpl.dispatchResume(FragmentManager.java:3012)
       at android.app.FragmentController.dispatchResume(FragmentController.java:205)
       at android.app.Activity.performResume(Activity.java:7647)
       at android.app.ActivityThread.performResumeActivity(ActivityThread.java:4018)
       at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:4058)
       at android.app.servertransaction.ResumeActivityItem.execute(ResumeActivityItem.java:51)
       at android.app.servertransaction.TransactionExecutor.executeLifecycleState(TransactionExecutor.java:145)
       at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:70)
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1960)
       at android.os.Handler.dispatchMessage(Handler.java:106)
       at android.os.Looper.loop(Looper.java:214)
       at android.app.ActivityThread.main(ActivityThread.java:7094)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:494)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:975)

### APK
Link to a Github repo where the bug is reproducible.

natario1, 2020-03-31:
Please update to latest version first. If the issue is still happening we'll fix it

florian-do, 2020-04-22:
Hello same problem here but only happen on my Samsung, work fine on a Pixel 3 and a Honor 8X 

### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: 2.6.2
- Camera engine used: camera2
- Reproducible in official demo app: no
- Device / Android version: *Samsung S7 (SM-G930W8), API 26*

### To Reproduce
While recording after 5, or 7 seconds

### Expected behavior
A clear and concise description of what you expected to happen.

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

```xml
<com.otaliastudios.cameraview.CameraView
	android:id="@+id/camera"
	android:layout_width="0dp"
	android:layout_height="0dp"
	android:layout_gravity="center"
	android:keepScreenOn="true"
	app:layout_constraintTop_toTopOf="parent"
	app:layout_constraintBottom_toBottomOf="parent"
	app:layout_constraintStart_toStartOf="parent"
	app:layout_constraintEnd_toEndOf="parent"
	app:cameraExperimental="false"
	app:cameraEngine="camera2"
	app:cameraPreview="glSurface"
	app:cameraPlaySounds="true"
	app:cameraGrid="off"
	app:cameraFlash="auto"
	app:cameraAudio="on"
	app:cameraVideoCodec="h264"
	app:cameraFacing="back"
	app:cameraGestureTap="autoFocus"
	app:cameraGestureLongTap="none"
	app:cameraGesturePinch="zoom"
	app:cameraGestureScrollHorizontal="none"
	app:cameraGestureScrollVertical="zoom"
	app:cameraAutoFocusMarker="@string/cameraview_default_autofocus_marker" />
```

### Logs
```
    Process: com.xxx.xxx.local, PID: 19303
    java.lang.RuntimeException: java.lang.StackOverflowError: stack size 1037KB
        at com.otaliastudios.cameraview.engine.CameraEngine$2.run(CameraEngine.java:239)
        at android.os.Handler.handleCallback(Handler.java:789)
        at android.os.Handler.dispatchMessage(Handler.java:98)
        at android.os.Looper.loop(Looper.java:164)
        at android.app.ActivityThread.main(ActivityThread.java:6944)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:327)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1374)
     Caused by: java.lang.StackOverflowError: stack size 1037KB
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.ensureToken(CameraOrchestrator.java:182)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.access$000(CameraOrchestrator.java:32)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:96)
        at com.google.android.gms.tasks.zzi.run(com.google.android.gms:play-services-tasks@@17.0.2:4)
        at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
        at com.otaliastudios.cameraview.internal.WorkerHandler$2.execute(WorkerHandler.java:112)
        at com.google.android.gms.tasks.zzj.zza(com.google.android.gms:play-services-tasks@@17.0.2:10)
        at com.google.android.gms.tasks.zzr.zza(com.google.android.gms:play-services-tasks@@17.0.2:19)
        at com.google.android.gms.tasks.zzu.zzb(com.google.android.gms:play-services-tasks@@17.0.2:112)
        at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(com.google.android.gms:play-services-tasks@@17.0.2:10)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2$1.onComplete(CameraOrchestrator.java:116)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
        at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.applyCompletionListener(CameraOrchestrator.java:192)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.access$100(CameraOrchestrator.java:32)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:101)
        at com.google.android.gms.tasks.zzi.run(com.google.android.gms:play-services-tasks@@17.0.2:4)
        at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
        at com.otaliastudios.cameraview.internal.WorkerHandler$2.execute(WorkerHandler.java:112)
        at com.google.android.gms.tasks.zzj.zza(com.google.android.gms:play-services-tasks@@17.0.2:10)
        at com.google.android.gms.tasks.zzr.zza(com.google.android.gms:play-services-tasks@@17.0.2:19)
        at com.google.android.gms.tasks.zzu.zzb(com.google.android.gms:play-services-tasks@@17.0.2:112)
        at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(com.google.android.gms:play-services-tasks@@17.0.2:10)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2$1.onComplete(CameraOrchestrator.java:116)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
        at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.applyCompletionListener(CameraOrchestrator.java:192)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.access$100(CameraOrchestrator.java:32)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:101)
        at com.google.android.gms.tasks.zzi.run(com.google.android.gms:play-services-tasks@@17.0.2:4)
        at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
        at com.otaliastudios.cameraview.internal.WorkerHandler$2.execute(WorkerHandler.java:112)
```

closed 2020-05-20 by stale[bot]