natario1/CameraView #813: How to set preview scaleType?

titanseason, 2020-03-31:
### How do I?
When I use CameraView with:
```
    <com.otaliastudios.cameraview.CameraView
        android:id="@+id/ocr_camera_view"
        android:layout_width="match_parent"
        android:layout_height="match_parent"
        android:keepScreenOn="true"
        app:cameraAudio="off"
        app:cameraAutoFocusResetDelay="500"
        app:cameraFacing="back"
        app:cameraFlash="off"
        app:cameraGestureLongTap="takePicture"
        app:cameraGestureTap="autoFocus"
        app:cameraGrid="off"
        app:cameraHdr="off"
        app:cameraMode="video"
        app:cameraPlaySounds="false"
        app:cameraPreview="surface"
        app:cameraWhiteBalance="auto" />
```
The preview image are always not fill the screen and left black borders. And the borders are always at bottom or right.

How to set preview scaleType? Like `ImageView`'s `centerCrop` or `fitCenter`.

### Version used
CameraView version: 2.6.2

natario1, 2020-03-31:
Do not use app:cameraPreview="surface" . It can't crop the preview.

titanseason, 2020-03-31:
@natario1 Thanks

Here's the solution:
Preview scaleType like ImageView's `centerCrop`:
remove `app:cameraPreview="surface"` from the layout.xml

Preview scaleType like ImageView's `fitCenter`:
add `android:layout_gravity="center"` to the layout.xml

closed 2020-03-31 by titanseason