natario1/CameraView #817: Capture delay and Burst mode with CameraView

thib-rdr, 2020-04-03:
### How do I?
Hi, and thanks for this really useful lib.

I'm using this lib to capture High-res images, but on some devices, users are complaining about a long capture delay. Upon analysis, it turns out the delay between `takePicture` and the callback `onPictureTaken` varies greatly, even on the same device and with good lighting conditions. Also, the preview of the camera is frozen during this delay.

Several questions : 
- Did anyone using this lib manage to decrease this delay, and has been able to take several high-res pictures in quick succession ?
- Is there a plan to implement Camera2 Burst mode in this lib in the future ?

Here are the relevant parts of my code, in case someone spots a mistake : 

        this.captureButton.setOnClickListener(view12 -> {
            captureButtonPressed = System.currentTimeMillis();
            thumbnailContainerLoadingMask.setVisibility(View.VISIBLE);
            cameraView.takePicture();
            captureButton.setVisibility(GONE);
        });
        ...
        this.cameraView.addCameraListener(new CameraListener() {

            ...

            @Override
            public void onPictureTaken(@NonNull PictureResult picture) {
                if(captureButtonPressed != 0L && CameraFragment.this.getContext() != null) {
                    long captureReady = System.currentTimeMillis() - captureButtonPressed;
                    Log.d(TAG, "Photo taken in " + captureReady + "ms"); //This is how I time the picture capture duration;
                }

                final int orientation = getActivity().getResources().getConfiguration().orientation;

                new AsyncDecodeBitmap((Picture pic) -> {
                    captureButton.setVisibility(View.VISIBLE);
                    pictures.add(0, pic);
                    addThumbnail(pic, 0);
                    thumbnailContainerLoadingMask.setVisibility(GONE);
                    lock = false;
                }, picture, getActivity().getFilesDir().getPath(), orientation).execute();
            }
            ...
        });

### Version used
I'm using CameraView v2.6.2

qingshanchangyi, 2020-04-29:
你好，我最近也遇到过这个问题，就是拍摄高质量的延迟比较高，但是这个问题在一些中低端手机上反而不常见，主要出现在高端机上。这样对用户体验来说非常不友好。请问你最后有什么解决方案了吗？

qingshanchangyi, 2020-04-29:
Hello, I have also encountered this problem recently, that is, the delay of high-quality shooting is relatively high, but this problem is not common in some low-end mobile phones, mainly in high-end phones. This is very unfriendly to the user experience. Do you have any final solution?

thib-rdr, 2020-04-29:
@qingshanchangyi No, haven't found a solution yet...

qingshanchangyi, 2020-04-30:
Okay...thanks

closed 2020-05-29 by stale[bot]

thib-rdr, 2020-07-10:
Just FYI for anyone reading this, we ended up switching to [CameraX](https://developer.android.com/training/camerax), instead of this lib. While this is a very nice lib, we needed a more robust solution, and under active development.