natario1/CameraView #818: camera preview show cropped view

Miteshmakwana73, 2020-04-08:
### Describe the bug
camera not showing full preview it is showing half preview but captured image is showing full preview i don't understand why it is happening

 this is my code for cameraview

```
 <com.otaliastudios.cameraview.CameraView
        android:id="@+id/camera"
        android:layout_width="wrap_content"
        android:layout_height="0dp"
        android:layout_centerInParent="true"
        android:layout_gravity="center"
        android:keepScreenOn="true"
        app:cameraAudio="on"
        app:cameraFacing="back"
        app:cameraFlash="off"
        app:cameraCropOutput="false"
        app:cameraGestureLongTap="none"
        app:cameraGesturePinch="none"
        app:cameraGestureScrollHorizontal="none"
        app:cameraGestureScrollVertical="none"
        app:cameraGestureTap="focus"
        app:cameraGrid="off"
        app:cameraJpegQuality="100"
        app:cameraPlaySounds="true"
        app:cameraSessionType="picture"
        app:cameraVideoQuality="max720p"
        app:cameraWhiteBalance="auto"
        app:layout_constraintEnd_toEndOf="parent"
        app:layout_constraintStart_toStartOf="parent"
        app:layout_constraintBottom_toBottomOf="parent"
        app:layout_constraintTop_toBottomOf="@+id/view19">

    </com.otaliastudios.cameraview.CameraView>
```
this is the camera preview of camera
[image]
this is captured image preview 
[image]

you can see that captured image showing more than camera preview 
my requirement to show same preview as captured image

ZakroKhitalishvili, 2020-05-01:
Same issue

fjr619, 2020-05-06:
same issues, the aspect ratio between preview and capture is not same when preview is match parent

ZakroKhitalishvili, 2020-05-06:
I solved this issue by cropping to preview's aspect ratio
Here is code:
```
private static Bitmap cropToAspectRatio(Bitmap image, float aspectRatio) {
        int width = image.getWidth();
        int height = image.getHeight();
        float imageAspectRatio = (float) width / (float) height;

        if (imageAspectRatio == aspectRatio) {
            return image;
        }

        int newWidth = (int) (height * aspectRatio);
        if (newWidth < width) {
            int margin = (width - newWidth) / 2;
            return Bitmap.createBitmap(image, margin, 0, newWidth, height);
        } else {
            int newHeight = (int) (width / aspectRatio);
            int margin = (height - newHeight) / 2;
            return Bitmap.createBitmap(image, 0, margin, width, newHeight);
        }
    }
```

Miteshmakwana73, 2020-05-06:
@ZakroKhitalishvili 
just add 
`        app:cameraCropOutput="true"` 

in CameraView it will work fine

ZakroKhitalishvili, 2020-05-06:
@Miteshmakwana73 
That attribute is not supported anymore in v2.

According to docs:

> Both cameraCropOutput and setCropOutput() have been removed. This was an expensive operation that worked with pictures only. In v2, if you want your output to be cropped to match the view bounds, you will use the *snapshot() APIs .

But a produced snapshot is significantly reduced in quality.

ZakroKhitalishvili, 2020-05-06:
To sum up this is not a bug, but a feature.

We have three options:

- Downgrade to v1
- Using snapshot API
- Cropping manually

Miteshmakwana73, 2020-05-07:
@ZakroKhitalishvili thank you for this information i don't know about this but i still think they should provide better solution for this scenario

closed 2020-06-05 by stale[bot]