natario1/CameraView #819: cameraPictureSizeAspectRatio cameraVideoSizeAspectRatio not working on all devices

AlexInn, 2020-04-08:
### How do I?
Hi, sorry for disturbing you.
Congratulations for the library, it's wonderful!
I have a problem on Xiaomi devices, the camera preview as the picture/video that I make wont get the correct aspect ratio that I've set on the View ( f.e.: cameraVideoSizeAspectRatio="1:1").

It's a known bug or maybe is me?

Thanks a million for your attention.

### Version used
CameraView version: 2.6.2

natario1, 2020-04-08:
You can't be sure that the device will support 1:1 pictures/videos. I suggest making **CameraView** 1:1 (for example with ConstraintLayout) and using takePictureSnapshot and takeVideoSnapshot to get 1:1 pictures/videos.

AlexInn, 2020-04-09:
Perfect! Sorry for the waste of time was my error!

Thanks again!

closed 2020-04-09 by AlexInn