natario1/CameraView #820: onConfigureFailed! Session android.hardware.camera2.impl.CameraCaptureSessionImpl@392a36c

nxnyk, 2020-04-09:
java.lang.RuntimeException: onConfigureFailed! Session android.hardware.camera2.impl.CameraCaptureSessionImpl@392a36c
        at com.otaliastudios.cameraview.engine.Camera2Engine$4.onConfigureFailed(Camera2Engine.java:584)
        at android.hardware.camera2.impl.CallbackProxies$SessionStateCallbackProxy.lambda$onConfigureFailed$1(CallbackProxies.java:64)
        at android.hardware.camera2.impl.-$$Lambda$CallbackProxies$SessionStateCallbackProxy$gvbTsp9UPpKJAbdycdci_ZW5BeI.run(Unknown Source:4)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:201)
        at android.os.HandlerThread.run(HandlerThread.java:65)

nxnyk, 2020-04-09:
i have this  issue on Redmi 8A,i want to catch this exception,but i can not catch on  onCameraError ,

natario1, 2020-04-21:
See #607

closed 2020-04-21 by natario1