natario1/CameraView #824: After take a picture, frame process will not receive any frame.

Weakey, 2020-04-14:
Hi,
In Full1PictureRecorder.java, it stops the preview callback by set to null before takePicture.
The preview callback will be set back on onPictureTaken function but forget to re-add callback buffer.

Reference:
https://developer.android.com/reference/android/hardware/Camera
Camera.PreviewCallback: a callback object that receives a copy of the preview frame, or null to stop receiving callbacks and "clear the buffer queue".

natario1, 2020-04-21:
Thanks for reporting @Weakey ! Would you like to fix it? I think we should call FrameManager.release() when removing the callback, and FrameManager.setUp() when adding it back.

jeffreyfjohnson, 2020-06-08:
@natario1 I can put in a fix for this. In the meantime, I'm trying to prove something out, do you know of a workaround here?

natario1, 2020-06-15:
I don't know any workaround, sorry, it's really just an internal bug. Thanks for fixing it

larkonesvk, 2020-06-15:
Hi @natario1,
 when do you plan to release new version with this fix?

PR #877 (2020-07-11): Restart Camera1 Frame Processing After Taking a Picture - Fixes Issue 824
- Fixes #824 
- Tests: No tests existed for the only edited class, `Full1PictureRecorder`
- Docs updated: No, fixes behind the scenes bug

### Solution
Took solution from @natario1 's comment on issue, simply needed to re-setup the frame manager after capturing a photo
closed 2020-07-11 by natario1