natario1/CameraView #825: Camera is not initialized after granting permissions.

VolodymyrMachekhin, 2020-04-15:
First of all, many thanks for such a great lib. It is best so far and I will be very honored to help you in any ways as soon as I can.  Right now can help only with reporting this issue.

I am using camera inside fragment of `ViewPager` inside my `MainActivity`. So a user can open camera by swipe. `Lifecycleowner` is not set yet. I use own permissions handling and setting `Lifecycleowner` after permission is granted only.
1. When permision is not granted and user openes the app `MainActivity`. 
2. User opens any other activity and goes back. So `MainActivity` wend `onStop`, `onResume`.
3. On `MainActivity` user swipes to camera where he grants the permission.
4. `Fragment` is set as `Lifecycleowner`. `CameraView.open` was called. 

In the result, user sees black screen. Related logs can be found below. Trying to capture image results to a crash.
*important note* - ommiting step 2 removes the problem and camera works.
 
Also with same steps reproduced this issue https://github.com/natario1/CameraView/issues/762. 
 
From logger
```
app passed NULL surface
```
- CameraView version: *2.6.2*
- Camera engine used: *camera2*
- Reproducible in official demo app: *no*
- Device / Android version: *emulator Pixel2, API 29, Samsung Galaxy S10, API 29*

### XML layout
```
    <com.otaliastudios.cameraview.CameraView
        android:id="@+id/view_camera"
        android:layout_width="match_parent"
        android:layout_height="match_parent"
        app:cameraAutoFocusMarker=""
        app:cameraEngine="camera2"
        app:cameraExperimental="false"
        app:cameraFacing="back"
        app:cameraFilter="@string/cameraview_filter_none"
        app:cameraGestureLongTap="autoFocus"
        app:cameraGesturePinch="zoom"
        app:cameraGestureTap="autoFocus"
        app:cameraMode="picture"
        app:cameraPlaySounds="true"
        app:cameraPreview="glSurface"
        app:cameraUseDeviceOrientation="true" />
```
### Crash log
```
 Process: com.mewe.dev, PID: 30127
    java.lang.RuntimeException: takePicture failed
        at android.hardware.Camera.native_takePicture(Native Method)
        at android.hardware.Camera.takePicture(Camera.java:1564)
        at com.otaliastudios.cameraview.picture.Full1PictureRecorder.take(Full1PictureRecorder.java:43)
        at com.otaliastudios.cameraview.engine.Camera1Engine.onTakePicture(Camera1Engine.java:348)
        at com.otaliastudios.cameraview.engine.CameraBaseEngine$3.run(CameraBaseEngine.java:509)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$3.run(CameraStateOrchestrator.java:100)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:75)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:72)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:100)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:214)
        at android.os.HandlerThread.run(HandlerThread.java:67)
```

ritesh-karmare, 2020-04-20:
@VolodymyrMachekhin 
Maybe you must have missed **Storage** permission

VolodymyrMachekhin, 2020-04-20:
No, it is granted. Also camera works after restarting the app.

ritesh-karmare, 2020-04-20:
Strange

closed 2020-05-17 by stale[bot]

NicolasBuquet, 2020-10-23:
Same situation on my side, using CameraView 2.6.4 in Fragment : app passed NULL Surface

anggit97, 2022-04-21:
@VolodymyrMachekhin are you found the solution with view pager?