natario1/CameraView #827: Audio / Video not Sync recording with Snapshot

buntupana, 2020-04-15:
### Describe the bug
Sometimes in some devices I'm getting a big sync delay between audio and video when I record a video using snapshot. It doesn't always happen

- CameraView version: 2.6.2
- Camera engine used: default
- Reproducible in official demo app: yes

### To Reproduce
1. Record Video using `cameraView.takeVideoSnapshot()`

### Expected behavior
Video and Audio should be synced

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

```xml
        <com.otaliastudios.cameraview.CameraView
            android:id="@+id/cameraView"
            android:layout_width="0dp"
            android:layout_height="0dp"
            android:keepScreenOn="true"
            app:cameraPlaySounds="true"
            app:cameraRequestPermissions="false"
            app:layout_constraintBottom_toBottomOf="parent"
            app:layout_constraintDimensionRatio="1"
            app:layout_constraintEnd_toEndOf="parent"
            app:layout_constraintStart_toStartOf="parent"
            app:layout_constraintTop_toTopOf="parent"/>
```

khanhtonbsv, 2020-04-29:
I have the same bug on Pixel 3XL os 9

buntupana, 2020-04-29:
One user had the same issue in a Pixel 2 as well

closed 2020-05-27 by stale[bot]

AdrianLeeElder, 2020-06-13:
Noticed this on the Pixel 4

chitrang200889, 2020-07-09:
@buntupana @AdrianLeeElder Are you able to resolve the issue?

In my case, audio starts earlier(~200-500ms) than video, is that same with you?

AdrianLeeElder, 2020-07-09:
> @buntupana @AdrianLeeElder Are you able to resolve the issue?
> 
> In my case, audio starts earlier(~200-500ms) than video, is that same with you?

Unfortunately no. My audio starts after the video, not before.

buntupana, 2020-07-11:
> @buntupana @AdrianLeeElder Are you able to resolve the issue?
> 
> In my case, audio starts earlier(~200-500ms) than video, is that same with you?

Me neither, I had to avoid using video snapshot recording.