natario1/CameraView #828: Something went wrong. java.lang.RuntimeException: start failed

ArsalImam, 2020-04-19:
### Describe the bug
Hi, 
I am getting **RuntimeException** on calling method **takeVideo** method with the default configurations. 
- CameraView version: *2.6.2*
- Camera engine used: *both*
- Reproducible in official demo app: *yes*
- Device / Android version: *Moto E4 Plus, API 27*
#### Investigation by myself
After digging into found, I have found some results which causing this issue, 
at **prepareMediaRecorder File: FullVideoRecorder:112** recorder properties are going correct which are `stub.videoFrameRate = 30`, size=1280*720 but after processing size automatically reducing to 120*96 and videoFrameRate=22 (in my opinion, this is causing error)
after more digging into it, I found the above mentioned processing is happening at FullVideoRecorder:163 and these configurations are not supported by my deivce as mentioned in the logs,
`CameraSource: Requested frame rate (22) is not supported: 10,15,20,15,20,24,30,30`

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. mode=front
2. cameraView.takeVideo(file);
3. Exception raised

### Expected behavior
should start video recording.

### XML layout

```xml
<CameraView
    android:layout_width="match_parent"
    android:layout_height="match_parent">
</CameraView>
```

### Logs
`2020-04-20 03:18:20.188 333-1164/? E/CameraSource: Requested frame rate (22) is not supported: 10,15,20,15,20,24,30,30
2020-04-20 03:18:20.189 333-1164/? E/StagefrightRecorder: [setupMPEG4orWEBMRecording] setupMediaSource Fail err=-19
2020-04-20 03:18:20.191 25915-26011/com.raply E/MediaRecorder: start failed: -19
2020-04-20 03:18:20.192 25915-26011/com.raply E/MediaRecorder: stop called in an invalid state: 0
2020-04-20 03:18:20.197 25915-26011/com.raply E/CameraEngine: onVideoResult result is null: something went wrong. java.lang.RuntimeException: start failed.
    java.lang.RuntimeException: start failed.
        at android.media.MediaRecorder.start(Native Method)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:301)
        at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:80)
        at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo(Camera1Engine.java:392)
        at com.otaliastudios.cameraview.engine.CameraBaseEngine$5.run(CameraBaseEngine.java:591)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$3.run(CameraStateOrchestrator.java:100)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:75)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:72)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:100)
        at com.google.android.gms.tasks.zzj.run(Unknown Source)
        at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
        at com.otaliastudios.cameraview.internal.WorkerHandler$2.execute(WorkerHandler.java:112)
        at com.google.android.gms.tasks.zzi.onComplete(Unknown Source)
        at com.google.android.gms.tasks.zzr.zza(Unknown Source)
        at com.google.android.gms.tasks.zzu.trySetResult(Unknown Source)
        at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(Unknown Source)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2$1.onComplete(CameraOrchestrator.java:116)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
        at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.applyCompletionListener(CameraOrchestrator.java:192)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.access$100(CameraOrchestrator.java:32)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:101)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
        at android.os.Handler.handleCallback(Handler.java:836)
        at android.os.Handler.dispatchMessage(Handler.java:103)
        at android.os.Looper.loop(Looper.java:203)
        at android.os.HandlerThread.run(HandlerThread.java:61)
`

derohimat, 2020-04-27:
I got this issue also

closed 2020-05-24 by stale[bot]

ismaielsharif, 2020-07-07:
From where is it stable? inactivity doesn't mean bugs are stable