natario1/CameraView #829: Camera default orientation does not match device orientation

Advice-Dog, 2020-04-20:
### Describe the bug
Within our Camera Activity, we allow the user to take a photo, which is cropped and then displayed to the user to allow them to confirm or retake the photo. When the device is in landscape, but held flat, the image is returned rotated. If the user then angles their device slightly, it will then take the photo correctly.

- CameraView version: 2.6.1
- Device / Android version: Multiple across different Android versions

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Rotate device into landscape
2. Put device on flat surface
3. Open Camera Activity
4. Pick up device while keeping device flat
5. Take a photo
6. Photo is rotated incorrectly

### Expected behavior
The image result should be returned with the correct orientation.

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

```        
<com.otaliastudios.cameraview.CameraView
            android:id="@+id/camera"
            android:layout_width="match_parent"
            android:layout_height="match_parent"
            android:adjustViewBounds="true"
            app:cameraAudio="off"
            app:cameraFacing="back"
            app:cameraSnapshotMaxHeight="2048"
            app:cameraSnapshotMaxWidth="2048"
            app:cameraUseDeviceOrientation="false"
            app:layout_constraintBottom_toBottomOf="parent"
            app:layout_constraintEnd_toEndOf="parent"
            app:layout_constraintStart_toStartOf="parent"
            app:layout_constraintTop_toTopOf="parent" />
```

### Code

If you add a `CameraListener`, after the first time `onOrientationChanged` is called it will return the correct result.
```
camera.addCameraListener(object: CameraListener() {
    override fun onOrientationChanged(orientation: Int) {
        super.onOrientationChanged(orientation)
        Log.e("Camera", "Orientation changed to $orientation")
    }
})
```

natario1, 2020-04-21:
When the device is held flat, it is not landscape nor portrait, it's just flat. There's no way for us to know where the user is with respect to the device.
You can use app:cameraUseDeviceOrientation="false" to disable orientation and always get the picture based on the activity orientation.

natario1, 2020-04-21:
Then I think I do not understand the issue, sorry! Could you take a video of your phone (with another phone) while doing this?

Advice-Dog, 2020-04-21:
Sorry for the double comments, Github is having issues right now.

I've recorded it, and put a gif on Imur. https://imgur.com/a/7Z4xdjQ

natario1, 2020-04-22:
And this is with cameraUseDeviceOrientation="false" ? (you can change it from the demo app settings - usign XML might not work in the demo app)

Advice-Dog, 2020-04-22:
Yes, I just retested it. It was false in the settings.

I'm testing on a Nexus 7, but we've also reproduced this behaviour on a Pixel 4.

Advice-Dog, 2020-05-13:
@natario1 Were you able to reproduce this issue?

AdrianLeeElder, 2020-05-25:
> Yes, I just retested it. It was false in the settings.
> 
> I'm testing on a Nexus 7, but we've also reproduced this behaviour on a Pixel 4.

I was able to reproduce this on a Pixel 4. It doesn't respect the device orientation. Even in an Activity where the mode is forced to portrait in the manifest, when I pull the information from the video it says orientation "270".

closed 2020-06-22 by stale[bot]

ppnfk, 2020-06-25:
Hi @natario1, I have the same issue:
mine is a horizontal activity, and set cameraUseDeviceOrientation = false.

in this case, I found that camereview has no change to set deviceOrientation  by                 mCameraEngine.getAngles().setDeviceOrientation(deviceOrientation);
in its mCallback.onDeviceOrientationChanged when there is no OrientationChanged event triggered.

I was wondering if there is a good point to invoke mCameraEngine.getAngles().setDeviceOrientation(deviceOrientation) when starting it.

thanks, 
Vincent

BartlomiejFrankow, 2020-08-26:
I got exactly same issue as @Advice-Dog have and can't find way to simply set and lock camera orientation without depending on device orientation.

ppnfk, 2020-08-26:
I fixed it by downloading source and put 
`        mCallback.onDeviceOrientationChanged(0);
`
on the last line of OrientationHelper.enable(). and built an aar to use.

for your reference.