natario1/CameraView #837: The video is stretched when using takeVideoSnapshot to take a multiFilter video

developertry, 2020-04-30:
### How do I?
Hello, I use multiFilter in the demo project. The video I get when using takeVideoSnapshot is stretched. How can I modify it?
I found that the rotation is 0 in the result page of the video, but the rotation is 90 when using takeVideo.

### Version used
2.6.2

developertry, 2020-04-30:
when takeVideoSnapshot
[image]

the result
[image]

closed 2020-05-29 by stale[bot]

grigM, 2020-07-23:
Hi @natario1 ! 
thank you for your's amazing work on this library!
 
i'm also have some video stretching problems when using multiFilter with videoSnapshot.
Preview on screen - is ok , with proper aspect ration, but video result is not.

maybe the reason of this line number 248 of SnapshotVideoRecorder.java:
mCurrentFilter.setSize(mResult.size.getWidth(), mResult.size.getWidth());

i't would be great. if you can check this

natario1, 2020-07-23:
I think this is because of #875 (open) @grigM

phamluanhanu2010, 2021-10-05:
@natario1 Do you have any update for this issue? I use the newest version but it still stretched video after user multifiler. 
Please guide me fix it.