natario1/CameraView #839: Using glMapBufferRange fails after toggling camera facing from front to back

wsidortsov, 2020-05-05:
I have encountered some strange behavior while trying to implement an OpenGL camera filter, involving compute shaders. I want to grab a camera frame as an RGB buffer of floating points, process it on CPU and then render it with OpenGL. I inherit BaseFilter, setup my compute program and SSBO buffer, dispatch computation in onPreDraw(), and finally render camera frame on the surface.

Everything works fine, until I either start recording video OR toggle camera facing between front and back. No matter how long I run the filter, as soon as I toggle facing, I am no longer able to get the contents of my SSBO with glMapBufferRange. OpenGL call fails with error code GL_OUT_OF_MEMORY, which I believe stands for a random "failed to obtain buffer" error. 

Toggling between front and back BEFORE setting the filter has no effect - the filter will start running normally with the current facing, but fail immediately after the next toggle. Surprisingly, setting filter to NONE before toggling does not help: I can start my filter with front facing, then set filter to NONE, then toggle facing, then set my filter again, and it will fail. It does not matter if I re-use the instance of my filter OR create new one every time I set it to CameraView.

Some random observations:
1. This issue happens only with Samsung devices, confirmed on Galaxy S8, Galaxy S10 and Galaxy Note 10 Lite. This issue cannot be reproduced on Google Pixel 3
2. This issue cannot be reproduced when I modify my compute shader to not sample camera frame via samplerExternalOES. Everything works fine.
3. This issue cannot be reproduced when I modify my compute shader to not perform any output to SSBO. Everything works fine.

My compute shader looks like this (sample code to reproduce the issue):
```
    private fun getComputeShaderText(cx: Int, cy: Int, bind: Int) =
    """#version 310 es
    #extension GL_OES_EGL_image_external_essl3: enable
    precision mediump float;
    
    layout(local_size_x = 8, local_size_y = 8) in;
    layout(std430) buffer;
    layout(binding = 0) uniform samplerExternalOES in_data;
    layout(binding = ${bind}) buffer Input { float elements[]; } out_data;
    
    void main() {
        if (gl_GlobalInvocationID.x >= ${cx}u || gl_GlobalInvocationID.y >= ${cy}u) return;
    
        float u = float(gl_GlobalInvocationID.x) / $cx.0;
        float v = float(gl_GlobalInvocationID.y) / $cy.0;
        vec3 texColor  = texture(in_data, vec2(u,v)).rgb;
        
        uint index = gl_GlobalInvocationID.x +  ${cx}u * gl_GlobalInvocationID.y;
        out_data.elements[index] = texColor.r;
    }
    """

```

And here is the entire filter class, which can reproduce the issue (all error checks are omitted for clarity, except for the relevant one):
```
import android.opengl.GLES31
import android.util.Log
import com.otaliastudios.cameraview.filter.BaseFilter

class DummyCameraFilter : BaseFilter() {

    private val ssboDimX = 64
    private val ssboDimY = 64
    private val ssboSize = ssboDimX * ssboDimY * 4 /* size of float */
    private val ssboBind = 1 /* binding point */

    private var renderProgram: Int = -1
    private var computeShader: Int = -1
    private var computeProgram: Int = -1
    private val ssbo = IntArray(1)

    override fun onCreate(programHandle: Int) {
        super.onCreate(programHandle)

        // keep program handle:
        this.renderProgram = programHandle

        // create shader:
        computeShader = GLES31.glCreateShader(GLES31.GL_COMPUTE_SHADER)
        GLES31.glShaderSource(computeShader, getComputeShaderText(ssboDimX, ssboDimY, ssboBind))
        GLES31.glCompileShader(computeShader)

        computeProgram = GLES31.glCreateProgram()
        GLES31.glAttachShader(computeProgram, computeShader)
        GLES31.glLinkProgram(computeProgram)

        // create ssbo:
        GLES31.glGenBuffers(1, ssbo, 0)
        GLES31.glBindBuffer(GLES31.GL_SHADER_STORAGE_BUFFER, ssbo[0])
        GLES31.glBufferData(GLES31.GL_SHADER_STORAGE_BUFFER, ssboSize, null, GLES31.GL_STREAM_COPY)
    }

    override fun onDestroy() {
        GLES31.glDeleteBuffers(1, ssbo, 0)
        GLES31.glDeleteShader(computeShader)
        GLES31.glDeleteProgram(computeProgram)
        super.onDestroy()
    }

    override fun onPreDraw(timestampUs: Long, transformMatrix: FloatArray) {
        super.onPreDraw(timestampUs, transformMatrix)

        // compute:
        GLES31.glUseProgram(computeProgram)
        GLES31.glBindBufferRange(GLES31.GL_SHADER_STORAGE_BUFFER, ssboBind, ssbo[0], 0, ssboSize)
        GLES31.glDispatchCompute(64 / 8, 64 / 8, 1)
        GLES31.glMemoryBarrier(GLES31.GL_SHADER_STORAGE_BARRIER_BIT)

        // fetch data:
        GLES31.glBindBuffer(GLES31.GL_SHADER_STORAGE_BUFFER, ssbo[0])
        GLES31.glMapBufferRange(GLES31.GL_SHADER_STORAGE_BUFFER, 0, ssboSize, GLES31.GL_MAP_READ_BIT)
        GLES31.glUnmapBuffer(GLES31.GL_SHADER_STORAGE_BUFFER)

        if (GLES31.glGetError() != GLES31.GL_NO_ERROR)
        { Log.d("CameraView", "This starts failing after toggle facing!") }
    }

    override fun onDraw(timestampUs: Long) {
        GLES31.glUseProgram(renderProgram)
        super.onDraw(timestampUs)
    }

    override fun getFragmentShader()
        = createDefaultFragmentShader()
}

```

### Extra Info
- CameraView version: 2.6.2
- Camera engine used: camera2
- Device version: Samsung Galaxy S8, S10, Note 10 Lite

### XML layout
```
    <com.otaliastudios.cameraview.CameraView
        android:id="@+id/cameraPreviewView"
        android:keepScreenOn="true"
        android:layout_width="match_parent"
        android:layout_height="0dp"
        app:cameraMode="video"
        app:cameraFacing="front"
        app:cameraAudio="off"
        app:cameraEngine="camera2"
        app:cameraExperimental="true"
        app:cameraPreview="glSurface"
        app:cameraRequestPermissions="false"
        app:layout_constraintDimensionRatio="3:4"
        app:layout_constraintTop_toTopOf="parent" />

```

natario1, 2020-05-05:
I don't know why this happens, but you could try putting this filter in a `MultiFilter` after a `NoFilter`. In this case the input texture will not be external anymore, and you can use sampler2D instead of samplerExternalOES in your compute shader, which is more reliable.

Also are you sure you should use `texture()` instead of `texelFetch()` in your shader?

wsidortsov, 2020-05-05:
Thank you for the reply. I was considering adding a render pass to copy OES texture to frame buffer, and MultiFilter seems to do the trick in one line of code. Nice! I have implemented this workaround and can confirm it really works.

I really need `texture()` sampling, as I want to downsample camera frame to a specific size, and I would like to use interpolation and normalized texture coordinates, something that `texelFetch()` cannot handle. I am using the result of the compute shader as a tensor for the neural network, which expects certain size.

As for the issue itself, I've noticed one more thing. After toggling to back camera, my filter starts emitting GL_OUT_OF_MEMORY error, but if I ignore it and continue processing, I can recover by somehow handling pause / resume cycle of my app. Literally, I switch to another app, then return back to mine, and the filter starts processing frames without errors. I believe, it is connected to cameraView handling lifecycle by shutting down camera on pause and then re-opening it on resume (I provide proper lifecycleOwner for the cameraView). Looks like pause/resume leads to resetting camera state back to normal, while toggling facing leaves something in an erroneous state.

Anyway, thanks again for your help! It saved me quite a lot of time!

natario1, 2020-05-05:
Great! I think it's a GPU driver issue, for example this huge bug exists https://community.arm.com/developer/tools-software/graphics/f/discussions/9432/is-extension-gl_oes_egl_image_external_essl3-not-working-properly-in-compute-shader-on-mali-g71-gpu . So you simply can't trust compute shaders with external textures in Android as of now.

closed 2020-05-27 by wsidortsov