natario1/CameraView #841: Setting camera Flash to AUTO not working

BhavyaRattan, 2020-05-06:
### Describe the bug

I am using takePictureSnapshot() to capture a picture. When I set the flash to AUTO it never triggers the flash even in low light conditions where the device camera does trigger a flash if I click picture from device camera in same scenario. 

- CameraView version: *2.6.2*
- Camera engine used: *both*
- Reproducible in official demo app: *yes*
- Device / Android version: *Pixel 3a XL, API 28*

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Open camera view app
2. Click on edit option (bottom left)
3. Change flash to `auto` by selecting from drop down spinner
4. Tap on `SNAP` button in extremely low light condition to capture a picture snapshot

### Expected behavior
In low light condition, flash mode `AUTO` should behave in the same way like the default device camera is behaving

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

```xml
 <com.otaliastudios.cameraview.CameraView
        android:id="@+id/camera"
        android:layout_width="wrap_content"
        android:layout_height="wrap_content"
        android:layout_gravity="center"
        android:layout_marginBottom="88dp"
        android:keepScreenOn="true"
        app:cameraExperimental="true"
        app:cameraEngine="camera2"
        app:cameraPreview="glSurface"
        app:cameraPlaySounds="true"
        app:cameraGrid="off"
        app:cameraFlash="auto"
        app:cameraAudio="on"
        app:cameraFacing="back"
        app:cameraGestureTap="autoFocus"
        app:cameraGestureLongTap="none"
        app:cameraGesturePinch="zoom"
        app:cameraGestureScrollHorizontal="filterControl1"
        app:cameraGestureScrollVertical="exposureCorrection"
        app:cameraMode="picture"
        app:cameraAutoFocusMarker="@string/cameraview_default_autofocus_marker">
```

### APK
Reproducible in the camera view demo app

fjr619, 2020-05-06:
i think `takePictureSnapshot` didnt trigger flash at all. because i got same result when set flash to ON. For `takePicture` it will trigger the flash

BhavyaRattan, 2020-05-07:
takePictureSnapshot() does trigger flash when set to TORCH. If user selected mode ON, I changed it to mode to TORCH, added 1 sec delay before capture to get a flash in picture and it works. I can not do the same thing for mode AUTO as it is based on the light conditions .

fjr619, 2020-05-07:
hmm weird, in my app it didnt trigger flash at all even if i set to ON

closed 2020-06-05 by stale[bot]