natario1/CameraView #844: Capture frames without using a view

hmawla, 2020-05-07:
### How do I?
Are we able to capture frames using camera.addFrameProcessor() without the need to use a view.
What I'm trying to do is to use DeepAR library with CameraView. So, DeepAR requires frames to be fed to it to render on a SurfaceView so I won't be needing CameraView's surface view anymore since it would just use more resources.

Is there a way to do so?
Or at least if I am able to user CameraView's GLSurfaceView with DeepAR though I think it won't work...

Any help would be appreciated!

### Version used
CameraView 2.2.0

closed 2020-06-05 by stale[bot]

timtsj, 2021-03-01:
@hmawla  u solved it?

hmawla, 2021-03-01:
> @hmawla hi, u solved it?

Had to migrate to CameraX and use its native renderer.