natario1/CameraView #849: Microphone permission mandatory for Picture mode

tartacutza, 2020-05-11:
### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: *2.6.2*
- Camera engine used: *both*
- Reproducible in official demo app: *no*
- Device / Android version: *Pixel, API 28*

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. Initialise CameraView object programatically. 
2. setMode(Mode.PICTURE), setRequestPermissions(false), setAudio(Audio.OFF) (optional)
3. permission for Microphone is still requested even if there is no need for it.

### Expected behavior
Only Permission for Camera should be required on PICTURE mode.

### XML layout
I view is added programatically, because I do permissions check manually.

### Screenshots
If applicable, add screenshots to help explain your problem.

### Logs
Use `CameraLogger.setLogLevel(LEVEL_INFO)` to see all logs into LogCat.

Use `CameraLogger.registerLogger()` to export to file or crash reporting service.

### APK
Link to a Github repo where the bug is reproducible.

Namazed, 2020-05-12:
Hi, @tartacutza 
If you look at checkPermissions method, you see that if set Audio.OFF, permission for microphone shouldn't request.
You should call setAudio(Audio.OFF) before camera open.

tartacutza, 2020-05-12:
Hi @Namazed, I did that too. As is mentioned in the To Reproduce section :)

Namazed, 2020-05-12:
Hm, but I do it, that I wrote and it works for me.
Maybe you can share source code your cameraView initialisation and we will try find solution?

tartacutza, 2020-05-12:
` private void initCamera() {
        if (mCameraView == null) {
            mCameraView = new CameraView(instance);
            mCameraView.setLifecycleOwner(this);
            mCameraView.setKeepScreenOn(true);
            mCameraView.setRequestPermissions(false);
            mCameraView.setMode(Mode.PICTURE);
            mCameraView.setAudio(Audio.OFF);
            mCameraView.setFlash(Flash.AUTO);

            if (mImageCallback == null) {
                mImageCallback = new CameraListener() {
                    @Override
                    public void onPictureTaken(@NonNull PictureResult result) {
                        super.onPictureTaken(result);
                        setupImagePreview(result.getData());
                    }
                };
                mCameraView.addCameraListener(mImageCallback);
            }

        }

        cameraContainer.addView(mCameraView);

    }`

tartacutza, 2020-05-12:
`
 private void initCamera() {
        if (mCameraView == null) {
            mCameraView = new CameraView(instance);
            mCameraView.setLifecycleOwner(this);
            mCameraView.setKeepScreenOn(true);
            mCameraView.setRequestPermissions(false);
            mCameraView.setMode(Mode.PICTURE);
            mCameraView.setAudio(Audio.OFF);
            mCameraView.setFlash(Flash.AUTO);

            if (mImageCallback == null) {
                mImageCallback = new CameraListener() {
                    @Override
                    public void onPictureTaken(@NonNull PictureResult result) {
                        super.onPictureTaken(result);
                        setupImagePreview(result.getData());
                    }
                };
                mCameraView.addCameraListener(mImageCallback);
            }

        }

        cameraContainer.addView(mCameraView);

    }
`

Namazed, 2020-05-13:
@tartacutza Try moved these methods:
```java
mCameraView.setRequestPermissions(false);
mCameraView.setMode(Mode.PICTURE);
mCameraView.setAudio(Audio.OFF);
```
before
```java
mCameraView.setLifecycleOwner(this);
```
I don't know when you call method initCamera, but if you call this method after when your lifecycleOwner change state on onResume, your camera will start open.

tartacutza, 2020-05-13:
Yess!!! that solved it
Thanks  a lot!

closed 2020-05-13 by tartacutza

Namazed, 2020-05-13:
You are welcome, @tartacutza :)