natario1/CameraView #850: Crash when start camera

Namazed, 2020-05-12:
### Describe the bug
- CameraView version: 2.6.2
- Camera engine used: camera1
- Reproducible in official demo app: If has specific device
- Device / Android version: many devices (Samsung s4, j5, lenovo tab m10 and etc), different versions

### Expected behavior
I expect that my App won't crash, I think we need wrap these places in tryCatch and throw CameraException instead of RuntimeException.

### XML layout
I create view programmatically

### Logs
I have only stacktrace.

```
Fatal Exception: java.lang.RuntimeException: setParameters failed
       at android.hardware.Camera.native_setParameters(Camera.java)
       at android.hardware.Camera.setParameters(Camera.java:1940)
       at com.otaliastudios.cameraview.engine.Camera1Engine.applyZoom(Camera1Engine.java:633)
       at com.otaliastudios.cameraview.engine.Camera1Engine.applyAllParameters(Camera1Engine.java:472)
       at com.otaliastudios.cameraview.engine.Camera1Engine.onStartEngine(Camera1Engine.java:199)
       at com.otaliastudios.cameraview.engine.CameraEngine$5.call(CameraEngine.java:390)
       at com.otaliastudios.cameraview.engine.CameraEngine$5.call(CameraEngine.java:383)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:69)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:60)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:100)
       at com.google.android.gms.tasks.zzj.run(zzj.java:4)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
       at com.otaliastudios.cameraview.internal.WorkerHandler$2.execute(WorkerHandler.java:112)
       at com.google.android.gms.tasks.zzi.onComplete(zzi.java:10)
       at com.google.android.gms.tasks.zzr.zza(zzr.java:19)
       at com.google.android.gms.tasks.zzu.trySetResult(zzu.java:94)
       at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(TaskCompletionSource.java:10)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2$1.onComplete(CameraOrchestrator.java:116)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.applyCompletionListener(CameraOrchestrator.java:192)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.access$100(CameraOrchestrator.java:32)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:101)
...........

```

```
Fatal Exception: java.lang.RuntimeException: getParameters failed (empty parameters)
       at android.hardware.Camera.native_getParameters(Camera.java)
       at android.hardware.Camera.getParameters(Camera.java:1953)
       at com.otaliastudios.cameraview.engine.Camera1Engine.getPreviewStreamAvailableSizes(Camera1Engine.java:103)
       at com.otaliastudios.cameraview.engine.CameraBaseEngine.computePreviewStreamSize(CameraBaseEngine.java:875)
       at com.otaliastudios.cameraview.engine.Camera1Engine.onStartBind(Camera1Engine.java:226)
       at com.otaliastudios.cameraview.engine.CameraEngine$8.call(CameraEngine.java:469)
       at com.otaliastudios.cameraview.engine.CameraEngine$8.call(CameraEngine.java:465)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:69)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:60)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:100)
       at com.google.android.gms.tasks.zzj.run(zzj.java:4)
.........
```

```
Fatal Exception: java.lang.RuntimeException: getParameters failed (empty parameters)
       at android.hardware.Camera.native_getParameters(Camera.java)
       at android.hardware.Camera.getParameters(Camera.java:1910)
       at com.otaliastudios.cameraview.engine.Camera1Engine.onStartEngine(Camera1Engine.java:196)
       at com.otaliastudios.cameraview.engine.CameraEngine$5.call(CameraEngine.java:390)
       at com.otaliastudios.cameraview.engine.CameraEngine$5.call(CameraEngine.java:383)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:69)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:60)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:100)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
       at android.os.Handler.handleCallback(Handler.java:789)
       at android.os.Handler.dispatchMessage(Handler.java:98)
       at android.os.Looper.loop(Looper.java:164)
       at android.os.HandlerThread.run(HandlerThread.java:65)
```

Namazed, 2020-06-05:
up

rvhamed, 2020-06-07:
same here

```
Fatal Exception: java.lang.RuntimeException: getParameters failed (empty parameters)
       at android.hardware.Camera.native_getParameters(Camera.java)
       at android.hardware.Camera.getParameters(Camera.java:1953)
       at com.otaliastudios.cameraview.engine.Camera1Engine.getPreviewStreamAvailableSizes(Camera1Engine.java:103)
       at com.otaliastudios.cameraview.engine.CameraBaseEngine.computePreviewStreamSize(CameraBaseEngine.java:857)
       at com.otaliastudios.cameraview.engine.Camera1Engine.onStartBind(Camera1Engine.java:196)
       at com.otaliastudios.cameraview.engine.CameraEngine$8.call(CameraEngine.java:469)
       at com.otaliastudios.cameraview.engine.CameraEngine$8.call(CameraEngine.java:465)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:69)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:60)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:100)
       at com.google.android.gms.tasks.zzj.run(:4)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
       at com.otaliastudios.cameraview.internal.WorkerHandler$2.execute(WorkerHandler.java:112)
       at com.google.android.gms.tasks.zzi.onComplete(:10)
       at com.google.android.gms.tasks.zzr.zza(:19)
       at com.google.android.gms.tasks.zzu.trySetResult(:94)
       at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(:10)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2$1.onComplete(CameraOrchestrator.java:116)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.applyCompletionListener(CameraOrchestrator.java:192)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.access$100(CameraOrchestrator.java:32)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:101)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:145)
       at android.os.HandlerThread.run(HandlerThread.java:61)
```

PattaFeuFeu, 2020-06-08:
Same for me. 

According to Play Store crash logs, it happened on the following devices out of our user base:

* Galaxy S4 (jflte) – Android 5.0
* Moto X (1st Gen) (ghost) – Android 5.1

```
java.lang.RuntimeException: 
 
  at android.hardware.Camera.native_getParameters (Camera.java)
 
  at android.hardware.Camera.getParameters (Camera.java:2027)
 
  at com.otaliastudios.cameraview.engine.Camera1Engine.getPreviewStreamAvailableSizes (Camera1Engine.java:103)
 
  at com.otaliastudios.cameraview.engine.CameraBaseEngine.computePreviewStreamSize (CameraBaseEngine.java:857)
 
  at com.otaliastudios.cameraview.engine.Camera1Engine.onStartBind (Camera1Engine.java:196)
 
  at com.otaliastudios.cameraview.engine.CameraEngine$8.call (CameraEngine.java:469)
 
  at com.otaliastudios.cameraview.engine.CameraEngine$8.call (CameraEngine.java:465)
 
  at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call (CameraStateOrchestrator.java:69)
 
  at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call (CameraStateOrchestrator.java:60)
 
  at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete (CameraOrchestrator.java:100)
 
  at com.google.android.gms.tasks.zzj.run (zzj.java)
 
  at com.otaliastudios.cameraview.internal.WorkerHandler.run (WorkerHandler.java:137)
 
  at com.otaliastudios.cameraview.internal.WorkerHandler$2.execute (WorkerHandler.java:112)
 
  at com.google.android.gms.tasks.zzi.a (zzi.java)
 
  at com.google.android.gms.tasks.zzr.a (zzr.java)
 
  at com.google.android.gms.tasks.zzu.b (zzu.java)
 
  at com.google.android.gms.tasks.TaskCompletionSource.a (TaskCompletionSource.java)
 
  at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2$1.onComplete (CameraOrchestrator.java:116)
 
  at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run (CameraOrchestrator.java:195)
 
  at com.otaliastudios.cameraview.internal.WorkerHandler.run (WorkerHandler.java:137)
 
  at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.applyCompletionListener (CameraOrchestrator.java:192)
 
  at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.access$100 (CameraOrchestrator.java:32)
 
  at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete (CameraOrchestrator.java:101)
 
  at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run (CameraOrchestrator.java:195)
 
  at android.os.Handler.handleCallback (Handler.java:739)
 
  at android.os.Handler.dispatchMessage (Handler.java:95)
 
  at android.os.Looper.loop (Looper.java:135)
 
  at android.os.HandlerThread.run (HandlerThread.java:61)
```

Namazed, 2020-06-29:
up

closed 2020-07-11 by natario1

cuibg, 2020-10-12:
@natario1 @Namazed up，the error has appeared in the version 2.6.3！

PR #851 (2020-07-11): Wrapped some crashes happened when start camera
- Fixes: #850
- Tests: no
- Docs updated: no

### Solution
I wrapped in tryCatch these problem places.