natario1/CameraView #856: StackOverflowError

73k05, 2020-05-18:
### Describe the bug
- CameraView version: *com.otaliastudios:cameraview:2.6.2*
- Camera engine used: *engine 2*
- Reproducible in official demo app: *did not try*
- Device / Android version: * Android 9 - LG G7 ThinQ*

### To Reproduce
No idea, coming from community

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

```xml
  <com.otaliastudios.cameraview.CameraView
        android:id="@+id/camera_view"
        android:layout_width="wrap_content"
        android:layout_height="wrap_content"
        android:keepScreenOn = "true"
        android:layout_gravity="center"
        app:cameraFacing = "back"
        app:cameraFlash = "off"
        app:cameraGestureLongTap = "autoFocus"
        app:cameraGesturePinch = "zoom"
        app:cameraGestureTap = "autoFocus"
        app:cameraPictureFormat = "jpeg"
        app:cameraVideoCodec = "deviceDefault"
        app:cameraWhiteBalance = "auto"
        />
```

### Screenshots
[image]

### Logs
```
Fatal Exception: java.lang.RuntimeException: java.lang.StackOverflowError: stack size 1041KB
       at com.otaliastudios.cameraview.engine.CameraEngine$2.run(CameraEngine.java:239)
       at android.os.Handler.handleCallback(Handler.java:873)
       at android.os.Handler.dispatchMessage(Handler.java:99)
       at android.os.Looper.loop(Looper.java:193)
       at android.app.ActivityThread.main(ActivityThread.java:6854)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:860)
```
### APK
https://github.com/akshay2211/PixImagePicker?utm_source=android-arsenal.com&utm_medium=referral&utm_campaign=6935

Namazed, 2020-05-22:
@natario1 the same on Engine 1.

73k05, 2020-07-24:
Hi again,

same error, different origin in the code 
[image]

**Stack**
`
Caused by java.lang.StackOverflowError: stack size 1039KB
       at com.otaliastudios.cameraview.metering.MeteringRegion.compareTo(MeteringRegion.java:10)
       at java.util.ComparableTimSort.countRunAndMakeAscending(ComparableTimSort.java:320)
       at java.util.ComparableTimSort.sort(ComparableTimSort.java:188)
       at java.util.Arrays.sort(Arrays.java:1302)
       at java.util.Arrays.sort(Arrays.java:1485)
       at java.util.ArrayList.sort(ArrayList.java:1470)
       at java.util.Collections.sort(Collections.java:206)
       at java.util.Collections.sort(Collections.java:159)
       at com.otaliastudios.cameraview.metering.MeteringRegions.get(MeteringRegions.java:41)
       at com.otaliastudios.cameraview.engine.Camera1Engine$11.run(Camera1Engine.java:844)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$3.run(CameraStateOrchestrator.java:100)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:75)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:72)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:100)
       at com.google.android.gms.tasks.zzj.run(zzj.java:4)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
       at com.otaliastudios.cameraview.internal.WorkerHandler$2.execute(WorkerHandler.java:112)
       at com.google.android.gms.tasks.zzi.onComplete(zzi.java:10)
`

natario1, 2020-07-24:
I can't see any recursion in these traces?

73k05, 2020-07-24:
> I can't see any recursion in these traces?

You mean the error is different and I should open a new ticket for that?

natario1, 2020-07-24:
One ticket is fine, but are you sure you pasted **all** the stacktrace?

73k05, 2020-07-25:
Maybe not ^^
`
Fatal Exception: java.lang.RuntimeException
java.lang.StackOverflowError: stack size 1039KB
Fatal Exception: java.lang.RuntimeException: java.lang.StackOverflowError: stack size 1039KB
       at com.otaliastudios.cameraview.engine.CameraEngine$2.run(CameraEngine.java:239)
       at android.os.Handler.handleCallback(Handler.java:883)
       at android.os.Handler.dispatchMessage(Handler.java:100)
       at android.os.Looper.loop(Looper.java:237)
       at android.app.ActivityThread.main(ActivityThread.java:7811)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1076)
Caused by java.lang.StackOverflowError
stack size 1039KB
Caused by java.lang.StackOverflowError: stack size 1039KB
       at com.otaliastudios.cameraview.metering.MeteringRegion.compareTo(MeteringRegion.java:10)
       at java.util.ComparableTimSort.countRunAndMakeAscending(ComparableTimSort.java:320)
       at java.util.ComparableTimSort.sort(ComparableTimSort.java:188)
       at java.util.Arrays.sort(Arrays.java:1302)
       at java.util.Arrays.sort(Arrays.java:1485)
       at java.util.ArrayList.sort(ArrayList.java:1470)
       at java.util.Collections.sort(Collections.java:206)
       at java.util.Collections.sort(Collections.java:159)
       at com.otaliastudios.cameraview.metering.MeteringRegions.get(MeteringRegions.java:41)
       at com.otaliastudios.cameraview.engine.Camera1Engine$11.run(Camera1Engine.java:844)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$3.run(CameraStateOrchestrator.java:100)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:75)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:72)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:100)
       at com.google.android.gms.tasks.zzj.run(zzj.java:4)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
       at com.otaliastudios.cameraview.internal.WorkerHandler$2.execute(WorkerHandler.java:112)
       at com.google.android.gms.tasks.zzi.onComplete(zzi.java:10)
       at com.google.android.gms.tasks.zzr.zza(zzr.java:19)
       at com.google.android.gms.tasks.zzu.trySetResult(zzu.java:94)
       at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(TaskCompletionSource.java:10)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2$1.onComplete(CameraOrchestrator.java:116)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.applyCompletionListener(CameraOrchestrator.java:192)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.access$100(CameraOrchestrator.java:32)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:101)
       at com.google.android.gms.tasks.zzj.run(zzj.java:4)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
       at com.otaliastudios.cameraview.internal.WorkerHandler$2.execute(WorkerHandler.java:112)
       at com.google.android.gms.tasks.zzi.onComplete(zzi.java:10)
       at com.google.android.gms.tasks.zzr.zza(zzr.java:19)
       at com.google.android.gms.tasks.zzu.trySetResult(zzu.java:94)
       at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(TaskCompletionSource.java:10)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2$1.onComplete(CameraOrchestrator.java:116)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.applyCompletionListener(CameraOrchestrator.java:192)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.access$100(CameraOrchestrator.java:32)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:101)
       at com.google.android.gms.tasks.zzj.run(zzj.java:4)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
       at com.otaliastudios.cameraview.internal.WorkerHandler$2.execute(WorkerHandler.java:112)
       at com.google.android.gms.tasks.zzi.onComplete(zzi.java:10)
       at com.google.android.gms.tasks.zzr.zza(zzr.java:19)
       at com.google.android.gms.tasks.zzu.trySetResult(zzu.java:94)
       at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(TaskCompletionSource.java:10)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2$1.onComplete(CameraOrchestrator.java:116)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
[... cut ...]
       at io.fabric.sdk.android.services.concurrency.DependencyPriorityBlockingQueue.take(DependencyPriorityBlockingQueue.java:65)
       at io.fabric.sdk.android.services.concurrency.DependencyPriorityBlockingQueue.take(DependencyPriorityBlockingQueue.java:46)
       at java.util.concurrent.ThreadPoolExecutor.getTask(ThreadPoolExecutor.java:1092)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1152)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
       at java.lang.Thread.run(Thread.java:919)
`

commit 8207e676 (2020-09-02): Some bug fixes (#953)

* Kotlin 1.4.0, fixes #940

* Fix multifilter bug, fixes #875

* Fix arithmetic exception, fixes #895

* Update dependencies

* Demo app in Kotlin

* Avoid stackoverflows on zoom/ev, fixes #856

* Remove Gemfile.lock

* Try to fix build
closed 2020-09-02 by natario1