natario1/CameraView #865: takeVideo does not respect device orientation

AdrianLeeElder, 2020-05-26:
When taking a video using `takeVideo`, the device orientation is not respected. For example, a recording can be started in `PORTRAIT` MODE (degree 0, upright phone) with a selected resolution of 1296x804, but the resulting video comes out in `LANDSCAPE` mode (degree 270).
- CameraView version: 2.6.2
- Camera engine used: camera1/camera2
- Reproducible in official demo app: yes
- Device / Android version: Pixel 4 XL API Level 29 Android 10

### To Reproduce
1. Go to Video mode
2. Select video width of 1296, video height of 804
3. Take video in `PORTRAIT` mode (degree 0, upright)

### Expected behavior
If I take a video with my phone held upright (PORTRAIT, 0 degrees), the resulting video orientation should be PORTRAIT, 0 degree rotation.

### APK
Demo

AdrianLeeElder, 2020-05-26:
It's important to note that this works fine with snapshots.

AdrianLeeElder, 2020-05-27:
I think I just might be misunderstanding how all this works. I guess I'm trying to record landscape videos in portrait mode and vice versa, but this doesn't seem possible outside of snapshots.

natario1, 2020-05-27:
Leave CameraView layout width/height as `wrap_content` (at least one of them), then use `setVideoSize()` to choose a video size that is landscape (width > height). This second step is probably redundant because the default size is typically landscape

closed 2020-06-24 by stale[bot]