natario1/CameraView #867: Front Cam goes black when i starts recording video in pixel

Ssehgal29, 2020-05-28:
### Describe the bug
Please add a clear description of what the bug is, **and** fill the list below.
- CameraView version: *2.6.2*
- Camera engine used: *camera2*
- Device / Android version: *Pixel, API 28*

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

        cameraView.setPlaySounds(false); //this attribute is also not working.
        cameraView.setFlash(Flash.OFF);
        cameraView.setVideoCodec(VideoCodec.H_264);
        cameraView.setWhiteBalance(WhiteBalance.AUTO);
        cameraView.setHdr(Hdr.ON);
        cameraView.setAudio(Audio.ON);
        cameraView.setAudioBitRate(64000);
        cameraView.setVideoBitRate(690000);
        cameraView.setPreviewFrameRate(30F);
        cameraView.setMode(Mode.VIDEO);

buntupana, 2020-06-23:
Having same issue

buntupana, 2020-07-11:
@Ssehgal29  Got it work in Pixel 2 using camera2, did you set experimental = true?

jonathanlmarsh, 2020-07-30:
I'm experiencing the same issue on Pixel 2 and Pixel 3 XL (but not the Pixel 3 🤷) with the front camera. However, setting app:cameraEngine="camera2" and app:cameraExperimental="true" seems to solve the issue.

Ssehgal29, 2020-08-05:
thanks @buntupana  and @jonathanlmarsh  it works for me also.

closed 2020-09-04 by stale[bot]

Bielak2000, 2023-02-24:
Unfortunately, it still doesn't work for me, anyone have any ideas? I am using camera2.