natario1/CameraView #873: Flip camera horizontally

KiemDuong, 2020-06-03:
### Problem to be addressed
I make an app that allows user to flip the camera. Does this library support this feature?
[image]

PraveenOjha, 2020-10-25:
yes with a custom filter  like this 
transform matrix for rotation with 180 along x or y axis .

 public class BasicFilter extends BaseFilter {

    private final static String FRAGMENT_SHADER = "#extension GL_OES_EGL_image_external : require\n"
            + "precision mediump float;\n"
            + "varying vec2 "+DEFAULT_FRAGMENT_TEXTURE_COORDINATE_NAME+";\n"
            + "uniform samplerExternalOES sTexture;\n" + "void main() {\n"
            + "  vec4 color = texture2D(sTexture, "+DEFAULT_FRAGMENT_TEXTURE_COORDINATE_NAME+");\n"
             + "  gl_FragColor = vec4(colorR, colorG, colorB, color.a);\n"
            + "}\n";

    public BlackAndWhiteFilter() { }

    @NonNull

    @Override
    protected void onPreDraw(long timestampUs, float[] transformMatrix)
    {
        
          android.opengl.Matrix.rotateM(transformMatrix, 0,180, 0, 1,0);
        super.onPreDraw(timestampUs,transformMatrix);
    }

        @Override
    public String getFragmentShader() {
        return FRAGMENT_SHADER;
    }
}

closed 2020-11-25 by stale[bot]

Child0fTheSun, 2023-02-23:
> 

I tried your code and got an error

`E/EGL_emulation: eglQueryContext 32c0  EGL_BAD_ATTRIBUTE
E/EGL_emulation: tid 30297: eglQueryContext(2019): error 0x3004 (EGL_BAD_ATTRIBUTE)
E/AndroidRuntime: FATAL EXCEPTION: GLThread 1470
    Process: ru.samplecode.mirror, PID: 30258
    java.lang.RuntimeException: Could not compile shader 35632: 'ERROR: 0:7: 'colorR' : undeclared identifier
    ERROR: 0:7: 'colorG' : undeclared identifier
    ERROR: 0:7: 'colorB' : undeclared identifier
    ' source: #extension GL_OES_EGL_image_external : require
    precision mediump float;
    varying vec2 vTextureCoord;
    uniform samplerExternalOES sTexture;
    void main() {
      vec4 color = texture2D(sTexture, vTextureCoord);
      gl_FragColor = vec4(colorR, colorG, colorB, color.a);
    }
    
        at com.otaliastudios.opengl.program.GlShader$Companion.compile(GlShader.kt:25)
        at com.otaliastudios.opengl.program.GlShader$Companion.access$compile(GlShader.kt:14)
        at com.otaliastudios.opengl.program.GlShader.<init>(GlShader.kt:8)
        at com.otaliastudios.opengl.program.GlProgram$Companion.create(GlProgram.kt:97)
        at com.otaliastudios.opengl.program.GlProgram.create(Unknown Source:2)
        at com.otaliastudios.cameraview.internal.GlTextureDrawer.draw(GlTextureDrawer.java:75)
        at com.otaliastudios.cameraview.preview.GlCameraPreview$Renderer.onDrawFrame(GlCameraPreview.java:234)
        at android.opengl.GLSurfaceView$GLThread.guardedRun(GLSurfaceView.java:1573)
        at android.opengl.GLSurfaceView$GLThread.run(GLSurfaceView.java:1272)
`

Child0fTheSun, 2023-02-23:
```
public class CameraFlipEffect extends BaseFilter {
    @Override
    protected void onPreDraw(long timestampUs, @NonNull float[] transformMatrix) {
        android.opengl.Matrix.rotateM(transformMatrix, 0, 180, 0, 1, 0);
        android.opengl.Matrix.translateM(transformMatrix, 0, -1, 0, -1);
        super.onPreDraw(timestampUs, transformMatrix);
    }

    @NonNull
    @Override
    public String getFragmentShader() {
        return createDefaultFragmentShader();
    }
}
```