natario1/CameraView #878: how can i use lookuptablefilter

Jey1992, 2020-06-09:
### How do I?
i want use a lookuptable image to translate filter , but i fannally didnot find a way to use it, can you fix my problem?

javaerwjy, 2020-06-10:
//        GPUImageLookup64Filter gpuImageLookup64Filter = new GPUImageLookup64Filter();
//        gpuImageLookup64Filter.setParameter1(1);
//        gpuImageLookup64Filter.setLookupResource(R.drawable.lookup_64);
//        cameraView.setFilter(new MultiFilter(new NoFilter(), gpuImageLookup64Filter));

Jey1992, 2020-06-10:
there is no GPUImageLookup64Filter class find

Jey1992, 2020-06-15:
> // GPUImageLookup64Filter gpuImageLookup64Filter = new GPUImageLookup64Filter();
> // gpuImageLookup64Filter.setParameter1(1);
> // gpuImageLookup64Filter.setLookupResource(R.drawable.lookup_64);
> // cameraView.setFilter(new MultiFilter(new NoFilter(), gpuImageLookup64Filter));

can you show me you GPUImageLookup64Filter class , thanks

closed 2020-07-12 by stale[bot]