natario1/CameraView #885: How to use  CameraView library as .aar file or dependency module?

chitrang200889, 2020-06-14:
I need to modify the library a bit as mentioned in #862, #855, and #607. 

So after the changes in a clone of this project, I generated .aar file but it was crashing similar to #503 

```
java.lang.NoClassDefFoundError: Failed resolution of: Lcom/otaliastudios/opengl/texture/GlTexture;
        at com.otaliastudios.cameraview.internal.GlTextureDrawer.<init>(GlTextureDrawer.java:35)
        at com.otaliastudios.cameraview.preview.GlCameraPreview$Renderer.onSurfaceCreated(GlCameraPreview.java:147)
        at android.opengl.GLSurfaceView$GLThread.guardedRun(GLSurfaceView.java:1541)
        at android.opengl.GLSurfaceView$GLThread.run(GLSurfaceView.java:1272)
     Caused by: java.lang.ClassNotFoundException: Didn't find class "com.otaliastudios.opengl.texture.GlTexture" on path: DexPathList[[zip file "/data/app/madlipz.eigenuity.com-Tlb1HLmltCFCmunoMEtA9Q==/base.apk"],nativeLibraryDirectories=[/data/app/madlipz.eigenuity.com-Tlb1HLmltCFCmunoMEtA9Q==/lib/arm64, /data/app/madlipz.eigenuity.com-Tlb1HLmltCFCmunoMEtA9Q==/base.apk!/lib/arm64-v8a, /system/lib64, /system/product/lib64]]
        at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:196)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:379)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
        at com.otaliastudios.cameraview.internal.GlTextureDrawer.<init>(GlTextureDrawer.java:35) 
        at com.otaliastudios.cameraview.preview.GlCameraPreview$Renderer.onSurfaceCreated(GlCameraPreview.java:147) 
        at android.opengl.GLSurfaceView$GLThread.guardedRun(GLSurfaceView.java:1541) 
        at android.opengl.GLSurfaceView$GLThread.run(GLSurfaceView.java:1272) 
```

Another solution I tried was to add the library as a module dependency. However, it requires to either **add** or **comment**  a lot of code related to apply plugin: 'com.github.dcendents.android-maven', apply plugin: 'com.jfrog.bintray' and maven installer. In addition to that, it loses the track with this original git repo which will add more complexity with future release if and when I require to upgrade the library. 

Please do suggest a stable solution that can comply with your future releases.

natario1, 2020-06-15:
When you use an AAR file you have to add all the dependencies of the AAR project to your own project.

chitrang200889, 2020-06-17:
I am following https://stackoverflow.com/a/50267507/3598052 this solution to generate .aar file which generates debug and release aar files. 

Is there any difference in the final builds? Which one should I use?

And this https://stackoverflow.com/a/49663101/3598052 to add & compile aar file.

ppnfk, 2020-06-25:
@chitrang200889  after you add the aar into your project,  you could add the dependencies into your app's build.gradle. such as     **implementation 'com.otaliastudios.opengl:egloo:0.4.0'**

closed 2020-07-06 by chitrang200889