natario1/CameraView #886: Snapshot Video Record fails randomly while recording more than 5 minutes

bambati, 2020-06-14:
- **CameraView version**: Release v2.6.2 with latest latest Commits on May 24, 2020
- **Camera engine used:** camera2
- **Reproducible in official demo app**: Yes
- **Device / Android version**: Billion Capture Plus Android Version: 8.1.0 | API level 27

Wanted to test Demo app for long duration so changed below line:
camera.takeVideoSnapshot(new File(getFilesDir(), "video.mp4"), 18000000);
And rest of code is same.

Noticed demo app always crashed with same error but different time duration

**### Snapshot of logcat:**
06-13 20:16:55.963   856   856 I /system/bin/tombstoned: received crash request for pid 21739
06-13 20:16:55.964 22103 22103 I crash_dump64: performing dump of process 21739 (target tid = 21924)
06-13 20:16:55.965 22103 22103 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
06-13 20:16:55.965 22103 22103 F DEBUG   : Build fingerprint: 'Billion/HS117/rimoB:8.1.0/HS117INDURH-409/409:user/release-keys'
06-13 20:16:55.965 22103 22103 F DEBUG   : Revision: '0'
06-13 20:16:55.965 22103 22103 F DEBUG   : ABI: 'arm64'
06-13 20:16:55.965 22103 22103 F DEBUG   : pid: 21739, tid: 21924, name: VideoEncoder  >>> com.otaliastudios.cameraview.demo <<<
06-13 20:16:55.965 22103 22103 F DEBUG   : signal 6 (SIGABRT), code -6 (SI_TKILL), fault addr --------
06-13 20:16:55.975 22103 22103 F DEBUG   : Abort message: 'Cannot get a dirty matrix!'
06-13 20:16:55.975 22103 22103 F DEBUG   :     x0   0000000000000000  x1   00000000000055a4  x2   0000000000000006  x3   0000000000000008
06-13 20:16:55.975 22103 22103 F DEBUG   :     x4   0000000000000000  x5   0000000000000000  x6   0000000000000000  x7   7f7f7f7f7f7f	7f7f
06-13 20:16:55.975 22103 22103 F DEBUG   :     x8   0000000000000083  x9   003a3c92d6a42fd6  x10  0000000000000000  x11  0000000000000001
06-13 20:16:55.975 22103 22103 F DEBUG   :     x12  ffffffffffffffff  x13  ffffffffffffffff  x14  ff00000000000000  x15  ffffffffffffffff
06-13 20:16:55.975 22103 22103 F DEBUG   :     x16  0000006321b71fa8  x17  00000076d75bb598  x18  0000000000000000  x19  00000000000054eb
06-13 20:16:55.975 22103 22103 F DEBUG   :     x20  00000000000055a4  x21  000000763ae58800  x22  000000763af6f540  x23  000000001c8dd718
06-13 20:16:55.975 22103 22103 F DEBUG   :     x24  0000000013b7bbe0  x25  0000000000000000  x26  000000001c8dc3e0  x27  0000000000000000
06-13 20:16:55.975 22103 22103 F DEBUG   :     x28  0000000000000000  x29  0000007638afed80  x30  00000076d7570df8
06-13 20:16:55.975 22103 22103 F DEBUG   :     sp   0000007638afed40  pc   00000076d7570e14  pstate 0000000060000000

06-13 20:16:56.311 22103 22103 F DEBUG   :
06-13 20:16:56.311 22103 22103 F DEBUG   : backtrace:
06-13 20:16:56.311 22103 22103 F DEBUG   :     #00 pc 000000000001de14  /system/lib64/libc.so (abort+104)
06-13 20:16:56.311 22103 22103 F DEBUG   :     #01 pc 0000000000007f20  /system/lib64/liblog.so (__android_log_assert+304)
06-13 20:16:56.311 22103 22103 F DEBUG   :     #02 pc 00000000000f4bc4  /system/lib64/libandroid_runtime.so (android::android_view_RenderNode_getTransformMatrix(long, long)+120)
06-13 20:16:56.311 22103 22103 F DEBUG   :     #03 pc 0000000000b22dc8  /system/framework/arm64/boot-framework.oat (offset 0x617000) (android.graphics.Canvas.nConcat [DEDUPED]+72)
06-13 20:16:56.311 22103 22103 F DEBUG   :     #04 pc 0000000000089b00  /dev/ashmem/dalvik-jit-code-cache (deleted)
06-13 20:16:56.328   823 21790 I mm-camera: <MCT   >< INFO> 1021: mct_controller_handle_SOF_proc: (sofdelay, curr_sofdelay) = (0, 0) kptr 6375

### Expected behavior
I want to use snapshot record for longer duration minimum 30mins and maximum 8hours.

bambati, 2020-06-14:
Noticed same issue with "Release Version:2.6.2".

06-14 22:57:18.757 21879 21914 V CameraEngine: onImageAvailable: Image acquired, dispatching.
06-14 22:57:18.757 21879 21914 V CameraCallbacks: dispatchFrame: 1592155638757 processors: 1
06-14 22:57:18.758 21879 22036 V CameraCallbacks: dispatchFrame: executing. Passing 1592155638757 to processors.
06-14 22:57:18.758 21879 22036 V DemoApp : Frame delayMillis: 84 FPS: 11
06-14 22:57:18.758 21879 22036 V Frame   : Frame with time 1592155638757 is being released.
06-14 22:57:18.758 21879 21917 I SnapshotVideoRecorder: scheduling frame.
06-14 22:57:18.758 21879 21917 I SnapshotVideoRecorder: dispatching frame.
06-14 22:57:18.758 21879 21917 V Pool    : GET - Reusing recycled item. Pool - count:4, active:2, recycled:2
06-14 22:57:18.758 21879 21917 V MediaEncoderEngine: Passing event to encoders: frame
06-14 22:57:18.758 21879 21917 V MediaEncoder: VideoEncoder Notify was called. Posting. pendingEvents: 2
06-14 22:57:18.759 21879 21917 V MediaEncoder: AudioEncoder Notify was called. Posting. pendingEvents: 1
06-14 22:57:18.759 21879 22055 V MediaEncoder: AudioEncoder Notify was called. Executing. pendingEvents: 1
06-14 22:57:18.796 22276 22276 I crash_dump64: obtaining output fd from tombstoned, type: kDebuggerdTombstone
06-14 22:57:18.797   856   856 I /system/bin/tombstoned: received crash request for pid 21879
06-14 22:57:18.798 22276 22276 I crash_dump64: performing dump of process 21879 (target tid = 22054)
06-14 22:57:18.798 22276 22276 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
06-14 22:57:18.799 22276 22276 F DEBUG   : Build fingerprint: 'Billion/HS117/rimoB:8.1.0/HS117INDURH-409/409:user/release-keys'
06-14 22:57:18.799 22276 22276 F DEBUG   : Revision: '0'
06-14 22:57:18.799 22276 22276 F DEBUG   : ABI: 'arm64'
06-14 22:57:18.799 22276 22276 F DEBUG   : pid: 21879, tid: 22054, name: VideoEncoder  >>> com.otaliastudios.cameraview.demo <<<
06-14 22:57:18.799 22276 22276 F DEBUG   : signal 6 (SIGABRT), code -6 (SI_TKILL), fault addr --------
06-14 22:57:18.806 22276 22276 F DEBUG   : Abort message: 'Cannot get a dirty matrix!'
06-14 22:57:18.806 22276 22276 F DEBUG   :     x0   0000000000000000  x1   0000000000005626  x2   0000000000000006  x3   0000000000000008
06-14 22:57:18.806 22276 22276 F DEBUG   :     x4   0000000000000000  x5   0000000000000000  x6   0000000000000000  x7   7f7f7f7f7f7f7f7f
06-14 22:57:18.806 22276 22276 F DEBUG   :     x8   0000000000000083  x9   003a3c92d6a42fd6  x10  0000000000000000  x11  0000000000000001
06-14 22:57:18.806 22276 22276 F DEBUG   :     x12  ffffffffffffffff  x13  ffffffffffffffff  x14  ff00000000000000  x15  ffffffffffffffff
06-14 22:57:18.806 22276 22276 F DEBUG   :     x16  0000006321b71fa8  x17  00000076d75bb598  x18  0000000000000000  x19  0000000000005577
06-14 22:57:18.807 22276 22276 F DEBUG   :     x20  0000000000005626  x21  00000076535dd800  x22  000000001979f070  x23  0000000000000000
06-14 22:57:18.807 22276 22276 F DEBUG   :     x24  0000000000000001  x25  0000000000000000  x26  0000000000000000  x27  0000000000000000
06-14 22:57:18.807 22276 22276 F DEBUG   :     x28  0000000000000000  x29  0000007638efcd90  x30  00000076d7570df8
06-14 22:57:18.807 22276 22276 F DEBUG   :     sp   0000007638efcd50  pc   00000076d7570e14  pstate 0000000060000000
06-14 22:57:18.828   823 21979 I mm-camera: <MCT   >< INFO> 1105: mct_stream_map_frame_number: SOF 14457: Book-keeping (frame_num, ack_frame_id) = [15258, 14460] in list
06-14 22:57:18.829   823 21979 I mm-camera: <MCT   >< INFO> 1021: mct_controller_handle_SOF_proc: (sofdelay, curr_sofdelay) = (0, 0) kptr 14457
06-14 22:57:18.909   823 21979 I mm-camera: <MCT   >< INFO> 1021: mct_controller_handle_SOF_proc: (sofdelay, curr_sofdelay) = (0, 0) kptr 14458
06-14 22:57:18.989   823 21979 I mm-camera: <MCT   >< INFO> 1021: mct_controller_handle_SOF_proc: (sofdelay, curr_sofdelay) = (0, 0) kptr 14459
06-14 22:57:19.070   823 21979 I mm-camera: <MCT   >< INFO> 1021: mct_controller_handle_SOF_proc: (sofdelay, curr_sofdelay) = (0, 0) kptr 14460
06-14 22:57:19.136 22276 22276 F DEBUG   :
06-14 22:57:19.136 22276 22276 F DEBUG   : backtrace:
06-14 22:57:19.136 22276 22276 F DEBUG   :     #00 pc 000000000001de14  /system/lib64/libc.so (abort+104)
06-14 22:57:19.136 22276 22276 F DEBUG   :     #01 pc 0000000000007f20  /system/lib64/liblog.so (__android_log_assert+304)
06-14 22:57:19.136 22276 22276 F DEBUG   :     #02 pc 00000000000f4b48  /system/lib64/libandroid_runtime.so (android::android_view_RenderNode_hasIdentityMatrix(long)+116)
06-14 22:57:19.136 22276 22276 F DEBUG   :     #03 pc 0000000000b23060  /system/framework/arm64/boot-framework.oat (offset 0x617000) (android.graphics.Canvas.nIsOpaque [DEDUPED]+64)
06-14 22:57:19.136 22276 22276 F DEBUG   :     #04 pc 0000000000006320  /dev/ashmem/dalvik-jit-code-cache (deleted)

Please help us to understand this issue.

iamkdblue, 2020-07-09:
@bambati are you getting Snapshot Video clear, without any glitches? I am getting glitches. how can I solve that?

bambati, 2020-08-25:
> @bambati are you getting Snapshot Video clear, without any glitches? I am getting glitches. how can I solve that?

@iamkdblue I tested with latest codebase  here and i doesn't change any. But i can say that Testing Device(Mobile model) also play impotent role. Mine mid range 1080p supported mobile device.

bambati, 2020-08-25:
@natario1 i tried little further in this debug and noticed this issue with SLOW draw operations.
Able to reproduce Very Easily with "Samsung A9" device with full screen WebOverlay. 

Log snippet may be helpful for your further debugging at your end. And also OPEN to support to get additional LOG if require.

08-20 17:17:33.822  8639 10960 E AndroidRuntime: FATAL EXCEPTION: VideoEncoder
08-20 17:17:33.822  8639 10960 E AndroidRuntime: Process: com.vsoftlabs.cameraview, PID: 8639
08-20 17:17:33.822  8639 10960 E AndroidRuntime: java.lang.RuntimeException: Probable deadlock detected due to WebView API being called on incorrect thread while the UI thread is blocked.
08-20 17:17:33.822  8639 10960 E AndroidRuntime: 	at xA0.c(chromium-Monochrome.aab-stable-414712573:6)
08-20 17:17:33.822  8639 10960 E AndroidRuntime: 	at com.android.webview.chromium.WebViewChromium.computeScroll(chromium-Monochrome.aab-stable-414712573:7)
08-20 17:17:33.822  8639 10960 E AndroidRuntime: 	at android.webkit.WebView.computeScroll(WebView.java:2778)
08-20 17:17:33.822  8639 10960 E AndroidRuntime: 	at android.view.View.draw(View.java:21607)
08-20 17:17:33.822  8639 10960 E AndroidRuntime: 	at android.view.ViewGroup.drawChild(ViewGroup.java:4558)
08-20 17:17:33.822  8639 10960 E AndroidRuntime: 	at com.otaliastudios.cameraview.overlay.OverlayLayout.doDrawChild(OverlayLayout.java:168)
08-20 17:17:33.822  8639 10960 E AndroidRuntime: 	at com.otaliastudios.cameraview.overlay.OverlayLayout.drawChild(OverlayLayout.java:157)
08-20 17:17:33.822  8639 10960 E AndroidRuntime: 	at android.view.ViewGroup.dispatchDraw(ViewGroup.java:4333)
08-20 17:17:33.822  8639 10960 E AndroidRuntime: 	at com.otaliastudios.cameraview.overlay.OverlayLayout.drawOn(OverlayLayout.java:139)
08-20 17:17:33.822  8639 10960 E AndroidRuntime: 	at com.otaliastudios.cameraview.overlay.OverlayDrawer.draw(OverlayDrawer.java:70)
08-20 17:17:33.822  8639 10960 E AndroidRuntime: 	at com.otaliastudios.cameraview.video.encoding.TextureMediaEncoder.onFrame(TextureMediaEncoder.java:220)
08-20 17:17:33.822  8639 10960 E AndroidRuntime: 	at com.otaliastudios.cameraview.video.encoding.TextureMediaEncoder.onEvent(TextureMediaEncoder.java:146)
08-20 17:17:33.822  8639 10960 E AndroidRuntime: 	at com.otaliastudios.cameraview.video.encoding.MediaEncoder$3.run(MediaEncoder.java:247)
08-20 17:17:33.822  8639 10960 E AndroidRuntime: 	at android.os.Handler.handleCallback(Handler.java:873)
08-20 17:17:33.822  8639 10960 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:99)
08-20 17:17:33.822  8639 10960 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:214)
08-20 17:17:33.822  8639 10960 E AndroidRuntime: 	at android.os.HandlerThread.run(HandlerThread.java:65)
08-20 17:17:33.822  8639 10960 E AndroidRuntime: Caused by: java.util.concurrent.TimeoutException
08-20 17:17:33.822  8639 10960 E AndroidRuntime: 	at java.util.concurrent.FutureTask.get(FutureTask.java:206)
08-20 17:17:33.822  8639 10960 E AndroidRuntime: 	at xA0.c(chromium-Monochrome.aab-stable-414712573:4)

FMiscia, 2021-01-03:
Same issue, I got fatal exception randomly caused by the Video Encoder:
`A/libc: Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x72006500630075 in tid 8116 (VideoEncoder), pid 31830`

Any clues?

Everything goes fine until i get: `W/Looper: Slow Looper main: doFrame is 499ms late because of 1 msg` and then the fatal exception.

I've seen a lot of issue of this type among the closed ones, but any good help so far

FMiscia, 2021-07-05:
I was able to drastically reduce those random crashes by reducing the key frames frequency. I set the  MediaFormat.KEY_I_FRAME_INTERVAL to 5 instead of 1, which means a key_frame every 5 seconds instead of 1  second.
If you guys wants to try the fix, you can use this fork:
[https://github.com/FMiscia/CameraView/](https://github.com/FMiscia/CameraView/)

pierre-the-fighter-pilot, 2023-02-28:
I've been experiencing a similar issue of random stoppage in video snapshot recordings with overlays.  in my app, the CameraView is in a sub-activity, so the app doesn't fully crash.  It just goes back to the parent activity and loses whatever video was in the process of being recorded.  I'm using a slightly modified version of CameraView based on 2.7.2 with Sdk 33.

This is the call stack from the logcat I get at the time of the bug:

2023-02-27 14:02:58.883 16371-16371 AndroidRuntime          com.cyrp.hockeytime                  D  Shutting down VM
2023-02-27 14:02:58.905 16371-16544 AndroidRuntime          com.cyrp.hockeytime                  E  FATAL EXCEPTION: VideoEncoder
                                                                                                    Process: com.cyrp.hockeytime, PID: 16371
                                                                                                    java.lang.IllegalStateException: No recording in progress, forgot to call #beginRecording()?
                                                                                                    	at android.graphics.RenderNode.endRecording(RenderNode.java:431)
                                                                                                    	at android.view.View.updateDisplayListIfDirty(View.java:21672)
                                                                                                    	at android.view.ViewGroup.recreateChildDisplayList(ViewGroup.java:4588)
                                                                                                    	at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:4561)
                                                                                                    	at android.view.View.updateDisplayListIfDirty(View.java:21625)
                                                                                                    	at android.view.ViewGroup.recreateChildDisplayList(ViewGroup.java:4588)
                                                                                                    	at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:4561)
                                                                                                    	at android.view.View.updateDisplayListIfDirty(View.java:21625)
                                                                                                    	at android.view.View.draw(View.java:22534)
                                                                                                    	at android.view.ViewGroup.drawChild(ViewGroup.java:4608)
                                                                                                    	at com.otaliastudios.cameraview.overlay.OverlayLayout.doDrawChild(OverlayLayout.java:180)
                                                                                                    	at com.otaliastudios.cameraview.overlay.OverlayLayout.drawChild(OverlayLayout.java:169)
                                                                                                    	at android.view.ViewGroup.dispatchDraw(ViewGroup.java:4361)
                                                                                                    	at com.otaliastudios.cameraview.overlay.OverlayLayout.drawOn(OverlayLayout.java:151)
                                                                                                    	at com.otaliastudios.cameraview.overlay.OverlayDrawer.draw(OverlayDrawer.java:74)
                                                                                                    	at com.otaliastudios.cameraview.video.encoding.TextureMediaEncoder.onFrame(TextureMediaEncoder.java:218)
                                                                                                    	at com.otaliastudios.cameraview.video.encoding.TextureMediaEncoder.onEvent(TextureMediaEncoder.java:146)
                                                                                                    	at com.otaliastudios.cameraview.video.encoding.MediaEncoder$3.run(MediaEncoder.java:244)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:938)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:233)
                                                                                                    	at android.os.Looper.loop(Looper.java:344)
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:67)
2023-02-27 14:02:58.905 16371-16371 AndroidRuntime          com.cyrp.hockeytime                  E  FATAL EXCEPTION: main
                                                                                                    Process: com.cyrp.hockeytime, PID: 16371
                                                                                                    java.lang.IllegalStateException: Recording currently in progress - missing #endRecording() call?
[... cut ...]
    boolean doDrawChild(Canvas canvas, View child, long drawingTime) {
        try {
            return super.drawChild(canvas, child, drawingTime);
        } catch (IllegalStateException e) {
            Log.e("CameraView", e.toString());
            return false;
        }
    }

This  addresses the first problem, but the second exception still occurs.  Unfortunately, I see nothing in the call stack where to catch that exception (and hope to ignore the problem for that frame).  Would anyone have an idea?

Thanks,

Pierre

pierre-the-fighter-pilot, 2023-03-03:
I speculated that what is probably happening (at least in my case) is that at some random point in time, I am getting an onFrame event before the overlay of the previous frame has finished being rendered.

In TextureMediaEncoder.java, line 218 in version 2.7.2, I added a synchronized statement to ensure that only one frame of overlay can be drawn at any time:

```
        // 3. Do the same for overlays with their own rotation.
        if (mConfig.hasOverlay()) {
            synchronized (mConfig.overlayTarget) {
                mConfig.overlayDrawer.draw(mConfig.overlayTarget);
            }
            Matrix.translateM(mConfig.overlayDrawer.getTransform(),
                    0, 0.5F, 0.5F, 0);
            Matrix.rotateM(mConfig.overlayDrawer.getTransform(),
                    0, mConfig.overlayRotation, 0, 0, 1);
            Matrix.translateM(mConfig.overlayDrawer.getTransform(),
                    0, -0.5F, -0.5F, 0);
        }
```
This is by no mean a rigourous fix, but I have not experienced the bug with this mod in place.