natario1/CameraView #887: Bad performance in frame processing: low fps

oiabtt, 2020-06-17:
Problem:
when deal with ML work such as facing tracking, we need process frame as fast as possible and then pass result to other module.  But when I run the demo, I find that It only has about 10 FPS even if we didn't do anything in the process function.
Code:
`camera.addFrameProcessor(new FrameProcessor() {
	private long lastTime = System.currentTimeMillis();
	@Override
	public void process(@NonNull Frame frame) {
		long newTime = frame.getTime();
		long delay = newTime - lastTime;
		lastTime = newTime;
		LOG.v("Frame delayMillis:", delay, "FPS:", 1000 / delay);
	}
});
`
> V/DemoApp: Frame delayMillis: 83 FPS: 12

Its unacceptable in most of the realtime ML program.  So why It shows a super low FPS even without any process? What should I do to improve the performance?

natario1, 2020-06-17:
There are many frame processing options that you can tune to improve this (see docs as always) and you can also play with camera1 vs camera2 and see which is better. That said, real-time processing is hard to do on the CPU. (and these are CPU frames)

oiabtt, 2020-06-17:
@natario1 I know real-time processing is hard. But the problem is that I did not get enogh high fps before any processing, as you can see in the code. So why the function public void process() loop is so slow? And at the same time I find that real-filters is  realtime, so how can I get the fps as high as real-filters in process function?

natario1, 2020-06-17:
It's because Filters run on the GPU, while Frames need to copy image data in CPU memory at each frame.

oiabtt, 2020-06-19:
@natario1 So how could  we do face landmark task in CameraView framework? You can see the realtime result in [https://github.com/zeusees/HyperLandmark](url), there is a GIF. Any suggestion? BTW we only need small picture (such as 112*112) to calculate face landmark.

closed 2020-07-17 by stale[bot]