natario1/CameraView #892: Cannot record video on some devices (both camera1 and camera2)

MohsenSamet, 2020-06-20:
### Describe the bug
- CameraView version: *2.6.2*
- Camera engine used: *both*
- Reproducible in official demo app: *yes*
- Device / Android version: *Nokia 3 API 24, 26, 28, Moto C Plus, Moto E (4) Plus, and some lenovo and other devices*

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app: (in specified devices)
1. Go to 'Video Mode (engine is not important, exception on both)'
2. Start recording video
3. See error

### Logs
Exception For when engine is camera1:
```
2020-06-20 12:18:24.219 8617-8643/com.otaliastudios.cameraview.demo I/CameraCallbacks: dispatchError com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: start failed.
    com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: start failed.
        at com.otaliastudios.cameraview.engine.CameraBaseEngine.onVideoResult(CameraBaseEngine.java:670)
        at com.otaliastudios.cameraview.engine.Camera1Engine.onVideoResult(Camera1Engine.java:436)
        at com.otaliastudios.cameraview.video.VideoRecorder.dispatchResult(VideoRecorder.java:138)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStop(FullVideoRecorder.java:359)
        at com.otaliastudios.cameraview.video.VideoRecorder.stop(VideoRecorder.java:98)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:320)
        at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:80)
        at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo(Camera1Engine.java:394)
        at com.otaliastudios.cameraview.engine.CameraBaseEngine$5.run(CameraBaseEngine.java:605)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$3.run(CameraStateOrchestrator.java:100)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:75)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:72)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:100)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.os.HandlerThread.run(HandlerThread.java:65)
     Caused by: java.lang.RuntimeException: start failed.
        at android.media.MediaRecorder.start(Native Method)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:314)
        at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:80) 
        at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo(Camera1Engine.java:394) 
        at com.otaliastudios.cameraview.engine.CameraBaseEngine$5.run(CameraBaseEngine.java:605) 
        at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$3.run(CameraStateOrchestrator.java:100) 
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:75) 
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:72) 
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:100) 
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195) 
        at android.os.Handler.handleCallback(Handler.java:873) 
        at android.os.Handler.dispatchMessage(Handler.java:99) 
        at android.os.Looper.loop(Looper.java:193) 
        at android.os.HandlerThread.run(HandlerThread.java:65) 
```

Exception For when engine is camera1:
```
2020-06-20 12:16:38.117 8255-8255/com.otaliastudios.cameraview.demo I/CameraCallbacks: dispatchError com.otaliastudios.cameraview.CameraException: android.hardware.camera2.CameraAccessException: CAMERA_ERROR (3): submitRequestList - configured surface is abandoned.
    com.otaliastudios.cameraview.CameraException: android.hardware.camera2.CameraAccessException: CAMERA_ERROR (3): submitRequestList - configured surface is abandoned.
        at com.otaliastudios.cameraview.engine.Camera2Engine.applyRepeatingRequestBuilder(Camera2Engine.java:252)
        at com.otaliastudios.cameraview.engine.Camera2Engine.onStartPreview(Camera2Engine.java:618)
        at com.otaliastudios.cameraview.engine.CameraEngine$10.call(CameraEngine.java:522)
        at com.otaliastudios.cameraview.engine.CameraEngine$10.call(CameraEngine.java:519)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:69)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:60)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:100)
        at com.google.android.gms.tasks.zzj.run(Unknown Source:4)
        at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
        at com.otaliastudios.cameraview.internal.WorkerHandler$2.execute(WorkerHandler.java:112)
        at com.google.android.gms.tasks.zzi.onComplete(Unknown Source:10)
        at com.google.android.gms.tasks.zzr.zza(Unknown Source:19)
        at com.google.android.gms.tasks.zzu.trySetResult(Unknown Source:94)
        at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(Unknown Source:10)
[... 43 lines cut ...]
        at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137) 
        at com.otaliastudios.cameraview.internal.WorkerHandler$2.execute(WorkerHandler.java:112) 
        at com.google.android.gms.tasks.zzi.onComplete(Unknown Source:10) 
        at com.google.android.gms.tasks.zzr.zza(Unknown Source:19) 
        at com.google.android.gms.tasks.zzu.setResult(Unknown Source:86) 
        at com.google.android.gms.tasks.zze.onSuccess(Unknown Source:8) 
        at com.google.android.gms.tasks.zzn.run(Unknown Source:4) 
        at com.google.android.gms.tasks.zzt.execute(Unknown Source:2) 
        at com.google.android.gms.tasks.zzm.onComplete(Unknown Source:11) 
        at com.google.android.gms.tasks.zzr.zza(Unknown Source:19) 
        at com.google.android.gms.tasks.zzu.trySetResult(Unknown Source:94) 
        at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(Unknown Source:10) 
        at com.otaliastudios.cameraview.engine.Camera2Engine$4.onConfigured(Camera2Engine.java:577) 
        at android.hardware.camera2.impl.CallbackProxies$SessionStateCallbackProxy.lambda$onConfigured$0(CallbackProxies.java:53) 
        at android.hardware.camera2.impl.-$$Lambda$CallbackProxies$SessionStateCallbackProxy$soW0qC12Osypoky6AfL3P2-TeDw.run(Unknown Source:4) 
        at android.os.Handler.handleCallback(Handler.java:873) 
        at android.os.Handler.dispatchMessage(Handler.java:99) 
        at android.os.Looper.loop(Looper.java:193) 
        at android.os.HandlerThread.run(HandlerThread.java:65) 
2020-06-20 12:16:38.117 8255-8255/com.otaliastudios.cameraview.demo I/CameraCallbacks: Caused by: android.os.ServiceSpecificException: submitRequestList - configured surface is abandoned. (code -22)
        at android.hardware.camera2.legacy.LegacyCameraDevice.submitRequestList(LegacyCameraDevice.java:452)
        at android.hardware.camera2.legacy.CameraDeviceUserShim.submitRequestList(CameraDeviceUserShim.java:459)
        at android.hardware.camera2.impl.ICameraDeviceUserWrapper.submitRequestList(ICameraDeviceUserWrapper.java:84)
        	... 38 more
```

closed 2020-07-18 by stale[bot]

appach22, 2021-03-30:
I'm also experiencing this issue on Nokia 1 (TA-1047).
Here is more verbose log:

`03-30 21:33:18.576 13913 13913 I CameraOrchestrator: MODE - Scheduling.
03-30 21:33:18.580 13913 14253 I CameraEngine: getPreviewStreamAvailableSizes: [176x144, 320x240, 352x288, 480x320, 480x368, 480x480, 640x480, 720x480, 800x480, 800x600, 864x480, 960x540, 960x720, 1280x720, 1440x1080, 1920x1080, 1920x1088]
03-30 21:33:18.582 13913 14253 I CameraEngine: computePreviewStreamSize: targetRatio: 3:4 targetMinSize: 480x1239
03-30 21:33:18.587 13913 14253 I CameraEngine: computePreviewStreamSize: result: 1440x1080 flip: true
03-30 21:33:18.587 13913 14253 I CameraEngine: onStartBind: Returning
03-30 21:33:18.589 13913 14253 I CameraOrchestrator: ENGINE >> BIND - Finished.
03-30 21:33:18.592 13913 14253 I CameraOrchestrator: BIND >> PREVIEW - Executing.
03-30 21:33:18.592 13913 14253 I CameraEngine: onStartPreview Dispatching onCameraPreviewStreamSizeChanged.
03-30 21:33:18.593 13913 14253 I CameraCallbacks: onCameraPreviewStreamSizeChanged: posting a requestLayout call. Preview stream size: 1080x1440
03-30 21:33:18.595 13913 14253 I CameraPreview: setStreamSize: desiredW= 1080 desiredH= 1440
03-30 21:33:18.609 13913 13913 I CameraCallbacks: dispatchOnCameraOpened com.otaliastudios.cameraview.engine.options.Camera1Options@e5a8bb0
03-30 21:33:18.615 13913 14253 I CameraEngine: computeCaptureSize: result: 1920x2560 flip: true mode: PICTURE
03-30 21:33:18.628 13913 13913 I CameraView: onMeasure: requested dimensions are (480[AT_MOST]x1239[EXACTLY])
03-30 21:33:18.629 13913 13913 I CameraView: onMeasure: previewSize is (1080.0x1440.0)
03-30 21:33:18.632 13913 13913 I CameraView: onMeasure: one dimension was EXACTLY, another AT_MOST. We have TRIED to fit the aspect ratio, but it's not guaranteed. (480x1239)
03-30 21:33:18.670 13913 14253 I CameraEngine: onStartPreview Starting preview with startPreview().
03-30 21:33:18.988 13913 13913 I OverlayLayout: normal draw called.
03-30 21:33:19.122 13913 13913 I CameraView: onMeasure: requested dimensions are (480[AT_MOST]x1239[EXACTLY])
03-30 21:33:19.124 13913 13913 I CameraView: onMeasure: previewSize is (1080.0x1440.0)
03-30 21:33:19.125 13913 13913 I CameraView: onMeasure: one dimension was EXACTLY, another AT_MOST. We have TRIED to fit the aspect ratio, but it's not guaranteed. (480x1239)
03-30 21:33:19.184 13913 13913 I OverlayLayout: normal draw called.
03-30 21:33:19.235 13913 14089 W libEGL  : EGLNativeWindowType 0x94284008 disconnect failed
03-30 21:33:19.347 13913 13913 I zzc     : onFragmentStopped()
03-30 21:33:19.364 13913 14145 E BufferQueueProducer: [SurfaceTexture-0-13913-0] disconnect: not connected (req=1)
03-30 21:33:19.365 13913 14145 W libEGL  : EGLNativeWindowType 0x94013008 disconnect failed
03-30 21:33:19.371 13913 13913 I SupportMapFragment: onFragmentStopped()
03-30 21:33:19.373 13913 13913 I MapFragmentAdv: onFragmentStopped()
03-30 21:33:19.375 13913 13913 I FitAnAddSubfragment: onFragmentStopped()
03-30 21:33:19.376 13913 13913 I FitAnAddFragment: onFragmentStopped()
03-30 21:33:19.380 13913 13913 I SupportRequestManagerFragment: onFragmentStopped()
03-30 21:33:19.382 13913 13913 I MainActivity: activityIsResumed = false
03-30 21:33:19.384 13913 13913 I MainActivity: onStop
03-30 21:33:19.385 13913 13913 I MainActivity: onSaveInstanceState()
03-30 21:33:19.385 13913 13913 I MainActivity: activityIsResumed = false
03-30 21:33:19.416 13913 14253 I CameraEngine: onStartPreview Started preview.
03-30 21:33:19.418 13913 14253 I CameraOrchestrator: BIND >> PREVIEW - Finished.
03-30 21:33:19.420 13913 14253 I CameraOrchestrator: ENGINE >> BIND - Executing.
03-30 21:33:19.426 13913 14253 W CameraOrchestrator: ENGINE >> BIND - State mismatch, aborting. current: PREVIEW from: ENGINE to: BIND
03-30 21:33:19.433 13913 14253 I CameraOrchestrator: ENGINE >> BIND - Finished because ABORTED.
03-30 21:33:19.434 13913 13913 I zzc     : onFragmentSaveInstanceState()
03-30 21:33:19.436 13913 14253 I CameraOrchestrator: BIND >> PREVIEW - Executing.
03-30 21:33:19.442 13913 14253 W CameraOrchestrator: BIND >> PREVIEW - State mismatch, aborting. current: PREVIEW from: BIND to: PREVIEW
03-30 21:33:19.452 13913 14253 I CameraOrchestrator: BIND >> PREVIEW - Finished because ABORTED.
03-30 21:33:19.454 13913 13913 I SupportMapFragment: onFragmentSaveInstanceState()
03-30 21:33:19.460 13913 14253 I CameraOrchestrator: MODE - Executing.
03-30 21:33:19.461 13913 14253 I CameraEngine: RESTART: scheduled. State: PREVIEW
03-30 21:33:19.461 13913 13913 I MapFragmentAdv: onFragmentSaveInstanceState()
03-30 21:33:19.461 13913 14253 I CameraEngine: STOP: scheduled. State: PREVIEW
03-30 21:33:19.463 13913 14253 I CameraOrchestrator: PREVIEW << BIND - Scheduling.
03-30 21:33:19.464 13913 14253 I CameraOrchestrator: BIND << ENGINE - Scheduling.
03-30 21:33:19.466 13913 14253 I CameraOrchestrator: ENGINE << OFF - Scheduling.
03-30 21:33:19.469 13913 14253 I CameraEngine: START: scheduled. State: PREVIEW
03-30 21:33:19.470 13913 14253 I CameraOrchestrator: OFF >> ENGINE - Scheduling.
03-30 21:33:19.472 13913 13913 I FitAnAddSubfragment: onFragmentSaveInstanceState()
03-30 21:33:19.474 13913 14253 I CameraOrchestrator: ENGINE >> BIND - Scheduling.
03-30 21:33:19.476 13913 14253 I CameraOrchestrator: BIND >> PREVIEW - Scheduling.
03-30 21:33:19.477 13913 14253 I CameraOrchestrator: MODE - Finished.
03-30 21:33:19.482 13913 14253 I CameraOrchestrator: PREVIEW << BIND - Executing.
[... cut ...]
03-30 21:33:21.018 13913 14253 W FullVideoRecorder: 	at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.run(CameraOrchestrator.java:137)
03-30 21:33:21.018 13913 14253 W FullVideoRecorder: 	at android.os.Handler.handleCallback(Handler.java:873)
03-30 21:33:21.018 13913 14253 W FullVideoRecorder: 	at android.os.Handler.dispatchMessage(Handler.java:99)
03-30 21:33:21.018 13913 14253 W FullVideoRecorder: 	at android.os.Looper.loop(Looper.java:193)
03-30 21:33:21.018 13913 14253 W FullVideoRecorder: 	at android.os.HandlerThread.run(HandlerThread.java:65)
`

appach22, 2022-02-11:
@MohsenSameti 
Have you found a solution for this issue?