natario1/CameraView #909: 录制视频时出现两种异常情况，经常性录制完成，关闭页面，在打开页面录制，完成后在关闭，如此反复几次出现下述情况

SmithYim, 2020-07-09:
07-09 16:12:24.711 21619-22221/com.ftsh.wanjia E/MediaRecorder: stop failed: -1007
07-09 16:12:24.711 21619-22221/com.ftsh.wanjia I/MediaRecorder: release()
07-09 16:12:24.716 21619-22221/com.ftsh.wanjia E/CameraEngine: onVideoResult result is null: something went wrong. java.lang.RuntimeException: stop failed.
    java.lang.RuntimeException: stop failed.
        at android.media.MediaRecorder.stop(Native Method)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStop(FullVideoRecorder.java:318)
        at com.otaliastudios.cameraview.video.VideoRecorder.stop(VideoRecorder.java:98)
        at com.otaliastudios.cameraview.engine.CameraBaseEngine.onStopVideo(CameraBaseEngine.java:640)
        at com.otaliastudios.cameraview.engine.CameraBaseEngine$7.run(CameraBaseEngine.java:631)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:75)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:72)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:100)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
        at android.os.Handler.handleCallback(Handler.java:739)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:179)
        at android.os.HandlerThread.run(HandlerThread.java:61)

07-09 16:14:23.437 21619-22221/com.ftsh.wanjia E/MediaRecorder: start failed: -2147483648
07-09 16:14:23.437 21619-22221/com.ftsh.wanjia D/OppoMmKeyLogWriter: writeLogToPartition: mediarecorder fail, 201
07-09 16:14:23.437 21619-22221/com.ftsh.wanjia E/ExtendedMediaRecorder: stop called in an invalid state: 0
07-09 16:14:23.437 21619-22221/com.ftsh.wanjia I/MediaRecorder: release()
07-09 16:14:23.438 21619-24672/com.ftsh.wanjia D/OppoMmKeyLogWriter: log = mediarecorder fail
07-09 16:14:23.438 21619-24672/com.ftsh.wanjia D/OppoMmKeyLogWriter: issue = video_record_fail
07-09 16:14:23.438 21619-24672/com.ftsh.wanjia D/OppoMmKeyLogWriter: desc = ����Ƶ¼���쳣
07-09 16:14:23.439 21619-22221/com.ftsh.wanjia E/CameraEngine: onVideoResult result is null: something went wrong. java.lang.RuntimeException: start failed.
    java.lang.RuntimeException: start failed.
        at android.media.MediaRecorder.start(Native Method)
        at com.otaliastudios.cameraview.video.FullVideoRecorder.onStart(FullVideoRecorder.java:301)
        at com.otaliastudios.cameraview.video.VideoRecorder.start(VideoRecorder.java:80)
        at com.otaliastudios.cameraview.engine.Camera1Engine.onTakeVideo(Camera1Engine.java:392)
        at com.otaliastudios.cameraview.engine.CameraBaseEngine$5.run(CameraBaseEngine.java:591)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$3.run(CameraStateOrchestrator.java:100)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:75)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$1.call(CameraOrchestrator.java:72)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:100)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
        at android.os.Handler.handleCallback(Handler.java:739)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:179)
        at android.os.HandlerThread.run(HandlerThread.java:61)

cuibg, 2020-07-10:
@SmithYim ，you are a funny guy，the author does not understand Chinese!haaaa

nxnyk, 2020-07-10:
> you are a funny guy，the author does not understand Chinese!haaaa

hhhh，he need Google Translate

cuibg, 2020-07-13:
@nxnyk 你们项目现在也用的这个？

cuibg, 2020-07-13:
@nxnyk 有什么Bug没，基本使用中

nxnyk, 2020-07-13:
> @nxnyk 有什么Bug没，基本使用中

还好吧，我没有录制视频功能，用的拍照，camera2 不太行 有的机型有问题，用camera1 还好

cuibg, 2020-07-14:
@nxnyk 好的，谢谢哈

closed 2020-08-11 by stale[bot]

cuibg, 2020-09-18:
@nxnyk 有些手机会出现RuntimeException，是在Camera1中出现的，你怎么处理的？

coolxiao, 2022-08-05:
@nxnyk @cuibg 我也是出现这个  你们解决了吗

cuibg, 2022-08-05:
这个不挂这个开源库的问题，我是把源码拉下来自己改的，然后把异常捕获了