natario1/CameraView #910: FATAL EXCEPTION: CameraViewEngine

leandro1995, 2020-07-11:
Hello a few days ago I found this error that I have seen in different error reports that have closed it without resolving it is the following:

E/Camera: Error 1
E/CameraEngine: Internal Camera1 error. 1
E/AndroidRuntime: FATAL EXCEPTION: CameraViewEngine
    Process: com.seci.cliente.app, PID: 20088
    com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: Internal Camera1 error. 1
        at com.otaliastudios.cameraview.engine.Camera1Engine.onError(Camera1Engine.java:91)
        at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1250)
        at android.os.Handler.dispatchMessage(Handler.java:109)
        at android.os.Looper.loop(Looper.java:166)
        at android.os.HandlerThread.run(HandlerThread.java:65)
     Caused by: java.lang.RuntimeException: Internal Camera1 error. 1
        at com.otaliastudios.cameraview.engine.Camera1Engine.onError(Camera1Engine.java:81)
        at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1250) 
        at android.os.Handler.dispatchMessage(Handler.java:109) 
        at android.os.Looper.loop(Looper.java:166) 
        at android.os.HandlerThread.run(HandlerThread.java:65) 
E/CameraEngine: EXCEPTION: Handler thread is gone. Replacing.
E/CameraEngine: EXCEPTION: Scheduling on the crash handler...
E/CameraEngine: EXCEPTION: Got CameraException. Since it is unrecoverable, executing destroy(false).
D/Camera: app passed NULL surface
E/CameraEngine: onStopBind Could not release surface java.io.IOException: setPreviewTexture failed
    java.io.IOException: setPreviewTexture failed
        at android.hardware.Camera.setPreviewTexture(Native Method)
        at com.otaliastudios.cameraview.engine.Camera1Engine.onStopBind(Camera1Engine.java:284)
        at com.otaliastudios.cameraview.engine.CameraEngine$9.call(CameraEngine.java:487)
        at com.otaliastudios.cameraview.engine.CameraEngine$9.call(CameraEngine.java:484)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:69)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:60)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:100)
        at com.google.android.gms.tasks.zzj.run(Unknown Source:4)
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler.run(WorkerHandler.java:137)
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler$2.execute(WorkerHandler.java:112)
        at com.google.android.gms.tasks.zzi.onComplete(Unknown Source:10)
        at com.google.android.gms.tasks.zzr.zza(Unknown Source:19)
        at com.google.android.gms.tasks.zzu.trySetResult(Unknown Source:94)
        at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(Unknown Source:10)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2$1.onComplete(CameraOrchestrator.java:116)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler.run(WorkerHandler.java:137)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.applyCompletionListener(CameraOrchestrator.java:192)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.access$100(CameraOrchestrator.java:32)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:101)
        at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
        at android.os.Handler.handleCallback(Handler.java:809)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:166)
        at android.os.HandlerThread.run(HandlerThread.java:65)
E/CameraEngine: EXCEPTION: Got CameraException. Dispatching to callback.

this happened to me on a huawei phone and a xiaomi i don't know how to correct this error

Bondes87, 2020-08-06:
I have a similar problem.

Steps to reproduce the behavior:
1. Create two fragments (first with cameraview and second without cameraview)
2. Open the first fragment
3. Take a photo and then open the second fragment.
4. Return to the first fragment.

leandro1995, 2020-08-08:
How about doing the steps but it does not generate problems all normal but my report applies that I do not use a fragment but an activity for taking photos:

`
```kotlin
cameraBinding.cameraView.addCameraListener(object : CameraListener() {
            override fun onPictureTaken(result: PictureResult) {
                super.onPictureTaken(result)

                GlobalScope.launch(Dispatchers.Main) {
                    setResult(Activity.RESULT_OK, Intent().apply {
                        putExtra(
                            Setting.CAMERA_URL,
                            FileUtil.savePhoto(
                                activity = this@CameraActivity,
                                byteArray = result.data
                            )
                        )
                    })
                    finish()
                }
            }
        })
````

What happens and I have seen in other libraries is that with huawie phones not all some most of them do not recognize the library and do not allow the photo I made a launch only with colleagues from huawie and the problem applied with 90% while the others did not There was a problem in other libraries I noticed that you have an activator for huawie, sorry for not having more information about the error, only that log :(

closed 2020-09-05 by stale[bot]

behlsoft, 2020-12-10:
am facing the same issue in Redmi 8

devsontx, 2020-12-23:
Same here on Nokia 2.1, Android 10

openmotion, 2021-03-26:
same here, android 10 huawei, if i switch between two activity (main activity then cam activity, back and go again )

KangDo96, 2021-05-17:
same here, how can I stop to happening this issue?

`E/Camera: Error 1
E/CameraEngine: Internal Camera1 error. 1
E/AndroidRuntime: FATAL EXCEPTION: CameraViewEngine
    Process: com.sportsyap.sportsyap, PID: 16704
    com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: Internal Camera1 error. 1
        at com.otaliastudios.cameraview.engine.Camera1Engine.onError(Camera1Engine.java:92)
        at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1268)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:223)
        at android.os.HandlerThread.run(HandlerThread.java:67)
     Caused by: java.lang.RuntimeException: Internal Camera1 error. 1
        at com.otaliastudios.cameraview.engine.Camera1Engine.onError(Camera1Engine.java:82)
        at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1268) 
        at android.os.Handler.dispatchMessage(Handler.java:106) 
        at android.os.Looper.loop(Looper.java:223) 
        at android.os.HandlerThread.run(HandlerThread.java:67) 
E/CameraEngine: EXCEPTION: Handler thread is gone. Replacing.
E/CameraEngine: EXCEPTION: Scheduling on the crash handler...`

combusterf, 2021-09-27:
Same issue here. XP8800, Also on android 10. 

Unlike the earlier comment I have "fixed" it by *removing* the lifecycle listener approach and using Fragment.onResume/onPause/onDestroy instead (as done in the documentation). I did also observe that in my setup onDestroy was never called on the fragment, but there was a new view being created after using the backbutton which might be relevant for debugging.

Before the fix, using the home button and returning to the app would also restore the camera functionality.

LuongCuong244, 2022-11-22:
This code has worked for me.

```
@Override
    protected void onDestroy() {
        if(cameraView != null){
            cameraView.close();
            cameraView.destroy();
        }
        super.onDestroy();
    }
```

hellosagar, 2024-07-24:
Anyone got a solution to this issue?