natario1/CameraView #914: v2.6.3 some methods not visible

oneornone, 2020-07-13:
when update to v2.6.3, some methods not visible for us, such us _addCameraListener_ etc. lots of methods and code are obfuscation.
any other solutions? thanks~

natario1, 2020-07-13:
Sorry 😄 it should be fixed now.
Make sure you clear your gradle dependencies: https://stackoverflow.com/questions/13565082/how-can-i-force-gradle-to-redownload-dependencies .

The best method would be to navigate in $HOME/.gradle/caches/ , find CameraView 2.6.3 folder and delete it.

oneornone, 2020-07-14:
> Sorry  it should be fixed now.
> Before trying make sure you clear your gradle dependencies: https://stackoverflow.com/questions/13565082/how-can-i-force-gradle-to-redownload-dependencies .
> 
> The best method would be to navigate in $HOME/.gradle/caches/modules-2/files-2.1/ , find CameraView 2.6.3 folder and delete it.

thanks sincerely, thumbs up :100:

oneornone, 2020-07-16:
> Sorry  it should be fixed now.
> Before trying make sure you clear your gradle dependencies: https://stackoverflow.com/questions/13565082/how-can-i-force-gradle-to-redownload-dependencies .
>
> The best [...]

[image]
the build task is failed? that means the latest version v2.6.3 still build with [isMinifyEnabled =true]? thanks~