natario1/CameraView #915: Question about mapping camera frames

mtekeli, 2020-07-13:
I was using Qt to access camera frames and process them in the real time. Since frame handle type is gl on Android (see [here](https://bugreports.qt.io/plugins/servlet/mobile#issue/QTBUG-61605))  I had to first map each frame and actually process it which brings much overhead to be done on the main thread. However Qt doesn't let you map in the background thread where problems start. You get a sluggish UI on many devices.

While I was looking for this issue I came across this project. Is this also a problem here? If not how is CameraView solving this issue? I tried to figure out from the sources but I lost track since I'm not a Java developer.

Thank you in advance

natario1, 2020-07-13:
I don't know what Qt is, but with frame processors ( https://natario1.github.io/CameraView/docs/frame-processing ) you can get the frames directly from camera sensor to CPU memory, without the gl texture step.

mtekeli, 2020-07-13:
Qt is an open source cross platform application development framework (C++). 

Then I guess this is not a general Android issue, rather how Qt's internal camera implementation works. 

Thanks again.

closed 2020-07-13 by mtekeli