natario1/CameraView #916: Filter is crashing on few devices.

parekhpratik24, 2020-07-15:
First of all, thanks for providing this great library. I hope you will create one for ios as well.

I am logging one crash report here, I have used filter for camera preview (its only preview, not recording or capturing photo).
On some devices,  filters are crashing. I have used combination of filters.

Samsung Galaxy S10 (beyond1q), 7680MB RAM, Android 10
Samsung Galaxy A10e (a10e), 1792MB RAM, Android 10

java.lang.RuntimeException: 
  at com.otaliastudios.opengl.core.Egloo.checkGlError (Egloo.kt:34)
  at com.otaliastudios.opengl.program.GlTextureProgram.onPreDraw (GlTextureProgram.kt:91)
  at com.otaliastudios.cameraview.filter.BaseFilter.onPreDraw (BaseFilter.java:168)
  at com.otaliastudios.cameraview.filters.GammaFilter.onPreDraw (GammaFilter.java:87)
  at com.otaliastudios.cameraview.filter.BaseFilter.draw (BaseFilter.java:160)
  at com.otaliastudios.cameraview.filter.MultiFilter.draw (MultiFilter.java:239)
  at com.otaliastudios.cameraview.internal.GlTextureDrawer.draw (GlTextureDrawer.java:85)
  at com.otaliastudios.cameraview.preview.GlCameraPreview$Renderer.onDrawFrame (GlCameraPreview.java:233)
  at android.opengl.GLSurfaceView$GLThread.guardedRun (GLSurfaceView.java:1573)
  at android.opengl.GLSurfaceView$GLThread.run (GLSurfaceView.java:1272)

Samsung Galaxy A01 (a01q), 2048MB RAM, Android 10
java.lang.RuntimeException: 
  at com.otaliastudios.opengl.core.Egloo.checkGlError (Egloo.kt:34)
  at com.otaliastudios.opengl.program.GlTextureProgram.onPreDraw (GlTextureProgram.kt:91)
  at com.otaliastudios.cameraview.filter.BaseFilter.onPreDraw (BaseFilter.java:168)
  at com.otaliastudios.cameraview.filters.BrightnessFilter.onPreDraw (BrightnessFilter.java:90)
  at com.otaliastudios.cameraview.filter.BaseFilter.draw (BaseFilter.java:160)
  at com.otaliastudios.cameraview.filter.MultiFilter.draw (MultiFilter.java:239)
  at com.otaliastudios.cameraview.internal.GlTextureDrawer.draw (GlTextureDrawer.java:85)
  at com.otaliastudios.cameraview.preview.GlCameraPreview$Renderer.onDrawFrame (GlCameraPreview.java:233)
  at android.opengl.GLSurfaceView$GLThread.guardedRun (GLSurfaceView.java:1573)
  at android.opengl.GLSurfaceView$GLThread.run (GLSurfaceView.java:1272)

IF you need more information for troubleshooting, let me know.

parekhpratik24, 2020-07-19:
Just want to follow up on this

natario1, 2020-07-22:
I don't know what's happening, the error message is missing from your logs and you also didn't fill the bug information requested.
- Does it happen when you don't use MultiFilter?
- Does it happen if you go back to 2.6.1?

parekhpratik24, 2020-07-25:
Hi Team,
This issue came in production. I never faced this issue with all of my test devices even with 1GB RAM. (And even simulators with lower configuration.)
This error is coming in prod and I have shared only information available to me. I am uploading screenshots for list of devices for your reference. 

**Does it happen when you don't use MultiFilter?** For me it was working with and without Multi filter. However it seems like on given devices, Multi filter is not working.

**Does it happen if you go back to 2.6.1?**
- it is not reproducible on my devices so I can't go back to 2.6.1.

I know i don't have lot of information to share but if you could reproduce this issue on any of these devices then probably you can fix it. 

I would love it if I can just do 2.6.3 in my gradle and all of my problems are solved :) 
[image]
[image]
[image]

parekhpratik24, 2020-08-01:
Hi Team,

Got any hint for this issue? This issue continue to reproduce. I have done try catch for the filter code but still failing with the same error on above devices. I think it is totally device specific. 

Also have you created same API for swift? if yes, can you provide the link to it?

closed 2020-08-29 by stale[bot]

PraveenOjha, 2020-10-14:
This Issue is present in all version where you rewrote the api for opengl . 
It was not present in 2.3.1 for sure . 

Error during glDrawArrays end: glError 0x502: invalid operation
 com.otaliastudios.opengl.core.Egloo.checkGlError

I will post after investigating 2.6.1 
but it is present in 

  2.6.2 
com.otaliastudios.opengl.core.Egloo.checkGlError (Egloo.kt:34)
com.otaliastudios.opengl.draw.GlRect.draw (GlRect.kt:69)
com.otaliastudios.opengl.program.GlProgram.onDraw (GlProgram.kt:60)
com.otaliastudios.cameraview.filter.BaseFilter.onDraw (BaseFilter.java:173)
com.otaliastudios.cameraview.filter.BaseFilter.draw (BaseFilter.java:161)

and in 

2.6.4 
com.otaliastudios.opengl.core.Egloo.checkGlError (Egloo.kt:47)
com.otaliastudios.opengl.draw.GlRect.draw (GlRect.kt:72)

Many device are producing this same crash ,unable to produce on emulator or on any of my devices.

parekhpratik24, 2020-11-07:
Is this resolved? @PraveenOjha : there is no special code, it is just multifilter that crashes on some devices. 
Root cause is this line : AndroidRuntime: at com.otaliastudios.opengl.core.Egloo.checkGlError(Egloo.kt:34)

It works fine on all of my devices across various android versions from 6 - android 10 and also works on simulator. But crashes on above mentioned devices.