natario1/CameraView #921: RuntimeException OutOfMemoryError ByteBufferFrameManager

73k05, 2020-07-24:
### The bug
The bug I cannot reproduce yet occurs from my community

- CameraView version: *2.6.2*
- Camera engine used: *camera2*
- Reproducible in official demo app: *unknonw*
- Device / Android version: *LG K20V, API 24*
- I have read the [FAQ page](https://natario1.github.io/CameraView/about/faq): *yes*

### To Reproduce
Steps to reproduce the behavior, possibly in the demo app:
1. I can't reproduce myself

### Expected behavior
Not crashing maybe or handle error

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

```xml
<com.otaliastudios.cameraview.CameraView
        android:id="@+id/camera_view"
        android:layout_width="wrap_content"
        android:layout_height="wrap_content"
        android:keepScreenOn = "true"
        android:layout_gravity="center"
        app:cameraFacing = "back"
        app:cameraFlash = "off"
        app:cameraGestureLongTap = "autoFocus"
        app:cameraGesturePinch = "zoom"
        app:cameraGestureTap = "autoFocus"
        app:cameraPictureFormat = "jpeg"
        app:cameraVideoCodec = "deviceDefault"
        app:cameraWhiteBalance = "auto"
        />
```

### Screenshots
[image]

### Logs
`Fatal Exception: java.lang.RuntimeException: java.lang.OutOfMemoryError: Failed to allocate a 1382412 byte allocation with 408264 free bytes and 398KB until OOM
       at com.otaliastudios.cameraview.engine.CameraEngine$2.run(CameraEngine.java:239)
       at android.os.Handler.handleCallback(Handler.java:751)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:154)
       at android.app.ActivityThread.main(ActivityThread.java:6316)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:872)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:762)`

`Caused by java.lang.OutOfMemoryError: Failed to allocate a 1382412 byte allocation with 408264 free bytes and 398KB until OOM
       at com.otaliastudios.cameraview.frame.ByteBufferFrameManager.setUp(ByteBufferFrameManager.java:88)
       at com.otaliastudios.cameraview.engine.Camera1Engine.onStartPreview(Camera1Engine.java:235)
       at com.otaliastudios.cameraview.engine.CameraEngine$10.call(CameraEngine.java:521)
       at com.otaliastudios.cameraview.engine.CameraEngine$10.call(CameraEngine.java:518)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:69)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:60)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:100)
       at com.google.android.gms.tasks.zzj.run(zzj.java:4)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
       at com.otaliastudios.cameraview.internal.WorkerHandler$2.execute(WorkerHandler.java:112)
       at com.google.android.gms.tasks.zzi.onComplete(zzi.java:10)
       at com.google.android.gms.tasks.zzr.zza(zzr.java:19)
       at com.google.android.gms.tasks.zzu.trySetResult(zzu.java:94)
       at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(TaskCompletionSource.java:10)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2$1.onComplete(CameraOrchestrator.java:116)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.applyCompletionListener(CameraOrchestrator.java:192)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.access$100(CameraOrchestrator.java:32)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:101)
       at com.google.android.gms.tasks.zzj.run(zzj.java:4)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
       at com.otaliastudios.cameraview.internal.WorkerHandler$2.execute(WorkerHandler.java:112)
       at com.google.android.gms.tasks.zzi.onComplete(zzi.java:10)
       at com.google.android.gms.tasks.zzr.zza(zzr.java:19)
       at com.google.android.gms.tasks.zzu.trySetResult(zzu.java:94)
       at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(TaskCompletionSource.java:10)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2$1.onComplete(CameraOrchestrator.java:116)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.applyCompletionListener(CameraOrchestrator.java:192)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.access$100(CameraOrchestrator.java:32)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:101)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$4.run(CameraOrchestrator.java:195)
       at android.os.Handler.handleCallback(Handler.java:751)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:154)
       at android.os.HandlerThread.run(HandlerThread.java:61)`

natario1, 2020-07-24:
The only good solution here is to stop using camera1 , at least frame processing... or you can decrease the preview size with `setPreviewStreamSize`.

But the library could at least avoid the crash and stop frame processing in this case.

73k05, 2020-07-24:
Hey, thanks for the quick response how can I do that?

Columbiec, 2020-09-29:
Reproduced for me in 2.6.4 release