natario1/CameraView #923: Attempt to invoke virtual method 'void android.graphics.SurfaceTexture.setDefaultBufferSize(int, int)' on a null object reference

chitrang200889, 2020-07-26:
### Describe the bug
- CameraView version: 2.6.3
- Camera engine used: camera2
- Device / Android version: All API levels

### To Reproduce
Not able to reproduce as 100+ bugs are reported on Firebase crashalytics

### Expected behavior
Should not crash

### XML layout
Part of the XML layout with the CameraView declaration, so we can read its attributes.

```xml
                <com.otaliastudios.cameraview.CameraView
                    android:id="@+id/cv_uca_cameraview"
                    android:layout_width="130dp"
                    android:layout_height="130dp"
                    android:layout_alignParentTop="true"
                    android:layout_alignParentRight="true"
                    android:layout_margin="@dimen/cell_margin_tiny"
                    android:keepScreenOn="true"
                    android:visibility="gone"
                    app:cameraAudio="mono"
                    app:cameraAudioCodec="deviceDefault"
                    app:cameraEngine="camera2"
                    app:cameraExperimental="true"
                    app:cameraFacing="front"
                    app:cameraFlash="off"
                    app:cameraGestureLongTap="none"
                    app:cameraGesturePinch="none"
                    app:cameraGestureScrollHorizontal="none"
                    app:cameraGestureScrollVertical="none"
                    app:cameraGestureTap="none"
                    app:cameraGrid="off"
                    app:cameraHdr="off"
                    app:cameraMode="video"
                    app:cameraPlaySounds="false"
                    app:cameraPreview="glSurface"
                    app:cameraPreviewFrameRate="30"
                    app:cameraPreviewFrameRateExact="true"
                    app:cameraRequestPermissions="false"
                    app:cameraUseDeviceOrientation="false"
                    app:cameraVideoCodec="h264"
                    app:cameraWhiteBalance="auto" />
                 </CameraView>
```
### Logs

```
Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.graphics.SurfaceTexture.setDefaultBufferSize(int, int)' on a null object reference
       at com.otaliastudios.cameraview.engine.Camera2Engine.onStartBind(Camera2Engine.java:506)
       at com.otaliastudios.cameraview.engine.CameraEngine$8.call(CameraEngine.java:470)
       at com.otaliastudios.cameraview.engine.CameraEngine$8.call(CameraEngine.java:466)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:69)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:60)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.onComplete(CameraOrchestrator.java:100)
       at com.google.android.gms.tasks.zzj.run(com.google.android.gms:play-services-tasks@@17.1.0:4)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
       at com.otaliastudios.cameraview.internal.WorkerHandler$2.execute(WorkerHandler.java:112)
       at com.google.android.gms.tasks.zzi.zza(com.google.android.gms:play-services-tasks@@17.1.0:10)
       at com.google.android.gms.tasks.zzq.zza(com.google.android.gms:play-services-tasks@@17.1.0:19)
       at com.google.android.gms.tasks.zzu.zzb(com.google.android.gms:play-services-tasks@@17.1.0:112)
       at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(com.google.android.gms:play-services-tasks@@17.1.0:10)
       at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2$1.onComplete(CameraOrchestrator.java:116)
       at com.google.android.gms.tasks.zzj.run(com.google.android.gms:play-services-tasks@@17.1.0:4)
       at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
       at com.otaliastudios.cameraview.internal.WorkerHandler$2.execute(WorkerHandler.java:112)
       at com.google.android.gms.tasks.zzi.zza(com.google.android.gms:play-services-tasks@@17.1.0:10)
       at com.google.android.gms.tasks.zzq.zza(com.google.android.gms:play-services-tasks@@17.1.0:19)
       at com.google.android.gms.tasks.zzu.zza(com.google.android.gms:play-services-tasks@@17.1.0:104)
       at com.google.android.gms.tasks.zzd.onSuccess(com.google.android.gms:play-services-tasks@@17.1.0:8)
       at com.google.android.gms.tasks.zzn.run(com.google.android.gms:play-services-tasks@@17.1.0:4)
       at com.google.android.gms.tasks.zzt.execute(com.google.android.gms:play-services-tasks@@17.1.0:2)
       at com.google.android.gms.tasks.zzm.zza(com.google.android.gms:play-services-tasks@@17.1.0:11)
       at com.google.android.gms.tasks.zzq.zza(com.google.android.gms:play-services-tasks@@17.1.0:19)
       at com.google.android.gms.tasks.zzu.zzb(com.google.android.gms:play-services-tasks@@17.1.0:112)
       at com.google.android.gms.tasks.TaskCompletionSource.trySetResult(com.google.android.gms:play-services-tasks@@17.1.0:10)
       at com.otaliastudios.cameraview.engine.Camera2Engine$2.onOpened(Camera2Engine.java:427)
       at android.hardware.camera2.impl.CameraDeviceImpl$1.run(CameraDeviceImpl.java:134)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:148)
       at android.os.HandlerThread.run(HandlerThread.java:61)
```

usmanrana-ilsa, 2022-01-18:
I'm facing exactly the same issue. CameraView is inside a fragment of ViewPager2, when page is half swiped and user takes application to background then on resuming the application crashes.

apolloandroid, 2022-11-23:
I faced this issue. Can not reproduce, but got from Firebase crashalytics. My CameraView: `<com.otaliastudios.cameraview.CameraView
        android:id="@+id/cameraView"
        android:layout_width="0dp"
        android:layout_height="0dp"
        android:layout_marginBottom="8dp"
        android:adjustViewBounds="true"
        android:keepScreenOn="true"
        app:cameraMode="picture"
        app:cameraPictureSizeAspectRatio="16:9"
        app:cameraGestureTap="autoFocus"
        app:cameraEngine="camera2"
        app:cameraExperimental="true" />
`

freelancer400, 2023-02-02:
facing crashes on firebase crashlytics and on google play console. Didn't find any solution.

AndroidWithRossyn, 2024-09-15:
java.lang.NullPointerException: Attempt to invoke virtual method 'void android.graphics.SurfaceTexture.setDefaultBufferSize(int, int)' on a null object reference
                                                                	at com.otaliastudios.cameraview.engine.Camera2Engine.onStartBind(Camera2Engine.java:517)
                                                                	at com.otaliastudios.cameraview.engine.CameraEngine$8.call(CameraEngine.java:470)
                                                                	at com.otaliastudios.cameraview.engine.CameraEngine$8.call(CameraEngine.java:466)
                                                                	at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:69)
                                                                	at com.otaliastudios.cameraview.engine.orchestrator.CameraStateOrchestrator$2.call(CameraStateOrchestrator.java:60)
                                                                	at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$3.run(CameraOrchestrator.java:152)
                                                                	at com.otaliastudios.cameraview.internal.WorkerHandler.run(WorkerHandler.java:137)
                                                                	at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.execute(CameraOrchestrator.java:147)
                                                                	at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator.access$100(CameraOrchestrator.java:34)
                                                                	at com.otaliastudios.cameraview.engine.orchestrator.CameraOrchestrator$2.run(CameraOrchestrator.java:137)
                                                                	at android.os.Handler.handleCallback(Handler.java:958)
                                                                	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                	at android.os.Looper.loopOnce(Looper.java:257)
                                                                	at android.os.Looper.loop(Looper.java:368)
                                                                	at android.os.HandlerThread.run(HandlerThread.java:67)