natario1/CameraView #925: Rectangle in Camera Preview

Ehsan-Home, 2020-07-27:
Hi. Thank you for the cool library. Amazing so far!🚀

I was wondering to know if you support putting rectangle in-camera preview like the link below:
https://lh3.googleusercontent.com/K78yAKNNnX4V-5aIPmqvvlYVJU_pgCh65N_B-iZ0W8hSC0jgeFGh8TiUb6ZIEXjfmvDX=h900

Any help or idea is highly appreciated. 👍🏻

GuanacoDevs, 2020-07-27:
The way I do it is to place an Image View on top of CameraView, then for analyzing I take the measurements at runtime, crop the Bitmap and pass it to the analyzer, that approach has work fine for me personally for over 3 years, but maybe there is a better approach certainly.

closed 2020-08-24 by stale[bot]

Greatkingcj, 2020-08-31:
> The way I do it is to place an Image View on top of CameraView, then for analyzing I take the measurements at runtime, crop the Bitmap and pass it to the analyzer, that approach has work fine for me personally for over 3 years, but maybe there is a better approach certainly.

i have a problem when using this method.  i want a fullscreen cameraview, so use this xml below
`<com.otaliastudios.cameraview.CameraView
        android:id="@+id/camera"
        android:layout_width="match_parent"
        android:layout_height="match_parent"
        android:adjustViewBounds="true"
        android:keepScreenOn="true"
        app:cameraUseDeviceOrientation="false"
        app:cameraEngine="camera1"
        app:cameraPlaySounds="false"
        app:cameraAudio="off"
        app:layout_constraintStart_toStartOf="parent"
        app:layout_constraintTop_toTopOf="parent"/>`
when i use takePictureSnapshot to get a bitmap， this bitmap size is 1108 * 2304 but the screen size is 1108 * 2264，then  i pass a rect and crop the bitmap , the result is wrong.  Any help or idea is highly appreciated.

rahullohra2903, 2021-09-01:
Is there a way to record the video only in the rectangular area?